arendst/Tasmota #2355: DomoticzUpdateTimer broken with 5.12.0i

0n3-70uch, 2018-04-04:
Hi, the function DomoticzUpdateTimer is not working anymore. Just set DomoticzUpdateTimer to 60 or another number. There is no domoticz/in message.

Hardware: Sonoff Basic
Firmware Version: 5.12.0i

- [X] What hardware you are using: Sonoff Basic
- [X] Provide the output of command ``status 0``
- [X] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [X] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

arendst, 2018-04-05:
It still works for me if set to 30 seconds:
```
09:46:50 MQT: tele/wemos6/STATE = {"Time":"2018-04-05T09:46:50","Uptime":"0T00:00:14","Vcc":3.077,"POWER":"OFF","Wifi":{"AP":2,"SSId":"indebuurt2","RSSI":80,"APMac":"00:22:6B:FE:8E:20"}}
09:46:50 MQT: domoticz/in = {"idx":10102,"nvalue":0,"svalue":"21.0;43.3;1","Battery":47,"RSSI":7}
09:46:50 MQT: tele/wemos6/SENSOR = {"Time":"2018-04-05T09:46:50","BMP280":{"Temperature":21.7,"Pressure":1003.0},"SHT3X-0x44":{"Temperature":21.0,"Humidity":43.3},"TempUnit":"C"}
09:47:13 MQT: domoticz/in = {"idx":10101,"nvalue":0,"svalue":"","Battery":47,"RSSI":7}
09:47:43 MQT: domoticz/in = {"idx":10101,"nvalue":0,"svalue":"","Battery":47,"RSSI":7}
```
But be aware it doesn't work for sensors (never did) but relays only.

0n3-70uch, 2018-04-05:
Hi, I found the mistake. The device settings have been deleted. Unfortunately, I do not know why... the `CFG_HOLDER` is stil the same. And the device sends no status when `DomoticzIdx` is 0.

Greetings
Fabi

closed 2018-04-05 by 0n3-70uch