arendst/Tasmota #2356: OTA possible with 1M firmware ?

ascillato, 2018-04-04:
Hi, 

Found on the **Espressif** site this [PDF](https://www.espressif.com/sites/default/files/99c-esp8266_ota_upgrade_en_v1.6.pdf) that shows that for _esp8266 devices with 4MB flash_ is possible to flash firmwares over OTA of up to 1MB. It uses 1MB for program, 1MB for receiving the new program from OTA and also it leaves 2MB for SPIFF or data.

[image]

@arendst, is this possible?

Jason2866, 2018-04-04:
Yes, can be read in the Wiki

ascillato, 2018-04-04:
Hi, I could not find any information on using this 2M ( 1024KB + 1024KB ) Program Memory.

How I can tell the compiler to use this?

Jason2866, 2018-04-04:
Tasmota only supports 1M memory. The reason for that is, it is build and optimized for Sonoff devices.
Sonoff devices have only 1M

Jason2866, 2018-04-04:
You can try this. In no way supported from @arendst 
https://github.com/arendst/Sonoff-Tasmota/wiki/Howto:-Expand-Sonoff-Basic-to-4MB-Flash

ascillato, 2018-04-04:
Sonoff devices have only 1MB flash memory, but that memory can be changed to 4MB chip.
In that case, according to the Espressif [PDF](https://www.espressif.com/sites/default/files/99c-esp8266_ota_upgrade_en_v1.6.pdf), should be possible to upload by OTA 1MB firmware.
My question is if this can be done, and how?

reloxx13, 2018-04-05:
place in a 4mb chip and flash the 1mb firmware. what is your probem?

ascillato, 2018-04-05:
The issue is:

How to configure Arduino to tell the ESP8266 to use the 2MB Program Memory ( 1024K + 1024K + 2048K SPIFFS ) for OTA upgrades as descripted on the Espressif [PDF ](https://www.espressif.com/sites/default/files/99c-esp8266_ota_upgrade_en_v1.6.pdf)?

arendst, 2018-04-05:
This discussion has nothing to do with Tasmota but everything with the ESP/Arduino library so I suggest you ask them.

In short. It works as long as your device has 4M memory. It's standard ESP/Arduino procedure for a linker script using the 4M memory model. I see no problem here. Take a look at EspEasy as they are using the 1M OTA model too in some of their distributions.

closed 2018-04-05 by arendst