arendst/Tasmota #2357: SK6812 - Works in DMA mode but has issues in BitBang mode

TimelessNL, 2018-04-04:
**Tasmota firmware** : 5.12.0
**Hardware**: NodeMCU 1.0 (currently for testing, since it has easy access to all the pins)
**user_config** : 
```
#define USE_WS2812_CTYPE     3
#define WS2812_LEDS          2 //purely for testing
```

I've 2 versions of the SK6812 LED strip at this moment. 
1. RGBNW (IP20)(60LED/Meter)
2. RGBW (IP65)(30LED/Meter)
But there seems to be a problem with nr2.

**SK6812 - RGBNW (IP20)(60LED/Meter)** (total length 5m)
DMA Mode (TXpin) = **OK**
BitBang mode (TXpin) = **OK**

**SK6812 - RGBW (IP65)(30LED/Meter)** (total length 5m)
DMA Mode  (TXpin) = **OK**
BitBang mode  (TXpin) = **ISSUE**

Let me give some scenarios:
**NOTE**: Start point for all is: LED strip is OFF and COLOR=1 (a.k.a RED).

**scenario 1:**
1. Set brightness to 20% (randomly chosen).
1a. Strip turns on 100% and RGBW (all leds)
2. Set brightness to 15% (randomly chosen).
2a. Strip turns to 15% and Red (2 configured leds)
3. Set brightness to 20% (randomly chosen).
3a. Strip turns on 100% and RGBW  (all leds)
4. Set brightness to 40% (randomly chosen).
4a. Strip turns to 40% and Red (2 configured leds)

**scenario 2:**
1. Turn Strip on 
1a. LED strip turns on 100% RGBW (all leds)
2. Turn strip off.
2a. LED strip turns off  (all leds)
3. Set brightness to 15% (randomly chosen).
3a. Strip turns to 15% and Red (2 configured leds)
4. Turn strip off.
4a. LED strip turns on 100% RGBW (all leds)
5. Turn Strip on.
5a. Strip turns to 15% and Red (2 configured leds)

**scenario 3:**
1. Turn Strip on.
1a. LED strip turns on 100% RGBW (all leds)
2. Set brightness to 15% (randomly chosen).
2a. Strip turns to 15% and Red  (2 configured leds)
3. Turn strip off.
3a. LED strip turns on 100% RGBW (all leds)
4. Turn Strip on.
4a. Strip turns to 15% and Red (2 configured leds)

I have no idea why DMA mode works fine while BitBang does not. Also my other SK6812 strip works just fine in both modes.

Jason2866, 2018-04-04:
I think the problem is the led strip. The other you have works as expected.
Try a another one from a different manufactor

TimelessNL, 2018-04-04:
I wish I could, these are the only SK6812 strips I have, they are both from the same seller, don't know if they are from the same manufacturer though.

Of course I could buy another strip but that does not explain why this current strip works in DMA mode.
And the error scenarios are repeatable, so it's not just random. Which I find strange.

Jason2866, 2018-04-04:
Yes, the problem can be repeated. But to find it, it is neccersary to check where to search...

TimelessNL, 2018-04-06:
Update: I also tested another SK6812 RGBW 60Leds/Meter IP67 (5M total) from thesame seller which gave me thesame issues in BitBang mode, DMA mode worked fine. Based on that I assume my other SK6812 is fine only the chips behave differently.

Frogmore42, 2018-04-07:
The timing will be different using DMA vs bit-banging.  Some chips are more tolerant of variations than others.

scottdalton, 2018-04-07:
I just ran into this same issue today testing out an SK6812 strip.  I had identical behavior with both bitbang and DMA modes.  Thinking it was a timing issue, I tried rebuilding with the CPU frequency upped from 80MHz to 160MHz and the problem resolved.  You might give it a try and see if this fixes it for you as well?

TimelessNL, 2018-04-08:
@scottdalton: I will test 160Mhz and report the results. Normally a higher CPU freq will generate more heat. What about the ESP8266. Is it safe to double the CPU freq?

@Forgmore42: True, but wouldn't that give more inconsistand results? My tests results are always repeatable, the results aren't just random. A timing issue will most likely result in glitching I suppose.

Frogmore42, 2018-04-10:
@TimelessNL perhaps, perhaps not.  If you have access to an oscilloscope, you could put it on the lines and see the difference in the signal.