arendst/Tasmota #2358: SerialSend witc

metrastar, 2018-04-04:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

arendst, 2018-04-05:
Should work. Give it a try and let us know.

metrastar, 2018-04-09:
It doesn't works :(

I'm not really firm with c programming, need a little bit help.
A string like "01040000000271CB" could be send with MQTT Command SerialSend4.
SerialSend4 should split it into 01|04|00|00|00|02|71|CB and convert into ASCII.

I have test it with following code:

#include <stdio.h>
#include <string.h>

int hex_to_int(char c)
{
    if (c >= 97)
        c = c - 32;
    int first = c / 16 - 3;
    int second = c % 16;
    int result = first * 10 + second;
    if (result > 9) result--;
    return result;
}

int hex_to_ascii(char c, char d){
        int high = hex_to_int(c) * 16;
        int low = hex_to_int(d);
        return high+low;
}

void append(char* s, char c)
{
        int len = strlen(s);
        s[len] = c;
        s[len+1] = '\0';
}

int main(){
        const char* st = "01040000000271CB";
        char str[50];
        int length = strlen(st);
        int i;
        char buf = 0;
        for(i = 0; i < length; i++){
                if(i % 2 != 0){
                        sprintf(str,"%c", hex_to_ascii(buf, st[i]));
                        printf(str);
                }else{
                        buf = st[i];
                }
        }
}

str should be appended to string and then send to serialbridgeserial->write()

Can anybody help me to implement it ?