arendst/Tasmota #2363: sonoff hostname

andyd999, 2018-04-05:
If i change the hostname to say livingroom_light cmd hostname displays correctly but i cant ping the hostname or use the hostname in chrome browser.  what am I doing wrong?

RaymondMouthaan, 2018-04-05:
Have you tried <your_hostname>.local ?

andyd999, 2018-04-05:
Yeah that dossnt work either Raymond but thanks

Frogmore42, 2018-04-05:
Did it work before you changed the name?
It has never worked for me either, but I assume that is because I don't have a bonjur server.  It hasn't been enough of an issue for me to spend a lot of effort figuring out what is missing.

andyd999, 2018-04-09:
I already have itunes installed. If i add the .local e.g. livingroom_light.local it still doesn't work

bollitec, 2018-05-07:
only for your info regarding Apple Bonjour service: https://learn.adafruit.com/bonjour-zeroconf-networking-for-windows-and-linux/overview
... in short: Bonjour w/o install itunes 
- download apple itunes installer, 
- unzip the itunes[32|64]setup.exe, 
- look for Bonjour[32|64].msi and double click on it to install the service only

ascillato, 2018-06-02:
Hi @andyd999 

Could you manage to solve this issue?

andyd999, 2018-06-02:
No mate never resolved

Frogmore42, 2018-06-02:
Do you have bonjour running?
Without that it won't work.

closed 2018-08-01 by stale[bot]