arendst/Tasmota #2365: Sonoff RF Bridge Randomly sent learned codes every reboot

tensopiseth, 2018-04-05:
Please help 
i got problem sonoff rf bridge randomly sent out learned code..
really annoying when boot or power out..
any solution??

Jason2866, 2018-04-05:
Anything connected on Gpios?
Bad power supply?

tensopiseth, 2018-04-06:
hmm.. i don't do anything after flash tasmota and selected model type to sonoff bridge and reboot it sent out code from 1 to 10 ( in console )
maybe i need to type some command?

tensopiseth, 2018-04-06:
ohh and ofor power supply it use micro usb to power i use powerbank 5v 2.1A for testing.

tensopiseth, 2018-04-07:
Now my problem seem to be fix by up date to 5.12.0j-mod-1.18.10
thanks to @arendst

tensopiseth, 2018-04-07:
Hmm Sth to do with MQTT
after i enable MQTT it same as before 

15:45:32 MQT: tele/sonoff433/INFO3 = {"RestartReason":"Software/System restart"}
15:45:32 MQT: stat/sonoff433/RESULT = {"Command":"Unknown"}
15:45:32 MQT: stat/sonoff433/RESULT = {"Command":"Unknown"}
15:45:32 MQT: stat/sonoff433/RESULT = {"Command":"Unknown"}
15:45:32 MQT: stat/sonoff433/RESULT = {"RfKey1":"Learned sent"}
15:45:32 MQT: stat/sonoff433/RESULT = {"RfKey3":"Default sent"}
15:45:32 MQT: stat/sonoff433/RESULT = {"RfKey4":"Default sent"}
15:45:32 MQT: stat/sonoff433/RESULT = {"RfKey8":"Default sent"}
15:45:32 MQT: stat/sonoff433/RESULT = {"RfKey7":"Default sent"}
15:45:32 MQT: stat/sonoff433/RESULT = {"RfKey5":"Default sent"}
15:45:32 MQT: stat/sonoff433/RESULT = {"RfKey6":"Default sent"}
15:45:32 MQT: stat/sonoff433/RESULT = {"RfKey9":"Default sent"}
15:45:32 MQT: stat/sonoff433/RESULT = {"RfKey10":"Default sent"}

tensopiseth, 2018-05-28:
for some resion it has been fix on Sonoff-Tasmota 5.13.1a thanks to Theo Arends

closed 2018-05-28 by tensopiseth

1shkoda1, 2020-11-17:
Installed 9.1.0. Have the same issue