arendst/Tasmota #2369: trying to upload tasmota onto the node mcu but having trouble with the mqtt packet size ive tried to eddit it in note ++ and id tried to put it at 1000 can eny one please exsplane how to install it correctly

clifty100, 2018-04-05:
Make sure these boxes are checked before submitting your issue - Thank you!

- [node mcu ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

reloxx13, 2018-04-05:
learn the diffrent about title and description first.

Frogmore42, 2018-04-07:
You should start with loading a pre-built binary first, which is covered in the wiki.  Once you know how to do that you can move on to more advanced things, but only if you really need them.