arendst/Tasmota #2370: Sonoff SV and Garage Door status update

Dev0003, 2018-04-05:
Hi Guys
Im currently using a Raspberry PI3 B,  Sonoff SV and magnetic reed switch to control my Garage door. I do not get status update on open/close when making contact with reed switch. Status seems to get stuck in open position and stays there no matter the reed switch status. There is no problem activating the buttons to open and close for 1 second.
[image]

I cannot understand  where im going wrong , see configuration below:

#In Configuration.yaml
cover:
  platform: mqtt
  name: "garage"
  state_topic: "cmnd/2/POWER2"
  command_topic: "cmnd/sonoffsv/POWER"
  payload_open: "ON"
  payload_close: "ON"
  payload_stop: "ON"
  state_open: "ON"
  state_close: "OFF"
  optimistic: false
  retain: false

#In Groups:
   garage_doors:
    name: Garage Door
    view: no
    control: hidden
    icon: mdi:garage
    entities:
      - cover.garage

#In Customization:
  cover.garage:
   device_class: garage
   friendly_name: Garage Door

When i toggle the reed switch , my sonoff console reflects these:
21:42:28 MQT: cmnd/2/POWER2 = OFF
21:42:28 MQT: cmnd/2/POWER2 = ON
21:42:30 MQT: cmnd/2/POWER2 = OFF
21:42:32 MQT: cmnd/2/POWER2 = ON

Please Help!!

curzon01, 2018-04-06:
Tasmota works as expect with correct payloads for your topic cmnd/2/POWER2. What's the issue on Tasmota here?

Dev0003, 2018-04-06:
Hi , I cannot get the status update (open/close) on Home Assistant. As per the picture , its stuck in open position, even when a change status on reed switch.

curzon01, 2018-04-06:
https://community.home-assistant.io/

Jason2866, 2018-04-06:
Or use a real Home Automation System....
For example OpenHab

dhami220, 2018-04-08:
If your relay is turning on and off with reed switch, it is likely because one of the command in tosmota console has not been applied yet. I had similar issue when I followed the following video.
https://www.youtube.com/watch?v=QMepwpyjMCY

Now, I can't figure out what that command was. I searched through that video, but can't find it (thought it was in the video comments).

dhami220, 2018-04-08:
try putting the following command in the sonoff console 

switchtopic2 2

geckofish, 2018-04-13:
Does it always toggle OFF->ON when you open or close the reed switch?  If so, the problem is probably your switchmode.

Jimma321, 2018-04-22:
Try switchtopic2 2.  I saw that in the description, but kept typing switchmode2 2.

closed 2018-06-21 by stale[bot]