arendst/Tasmota #2376: Basic help with Serial Bridge on Wemos D1 required please

charlietomo, 2018-04-06:
I'm after some basic help in order to get the recent Serial Bridge working as I think I am missing something (or have a hardware issue). I'm using a Wemos D1 mini, set as 18 Generic. I have connected a TTL-RS232 (max232 based) board (RX/TX/VCC/GND) and connected this to my RS232 device, which sends out messages. I would like those messages to be sent over MQTT.

I have read the [manual](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#serial-bridge) but am unclear if I need to configure Wemos pins. Regardless, I have tried setting TX and RX to 67 SerBr TX and 68 SerBr RX (and the other way round) with no results. I have tried setting none, and no results. I can't find a way to force "hardware mode"?

I have set BaudRate with `Baudrate 9600`. I have tried with default and with `SerialLog 4` and am looking for outputs on the console when my device is sending messages,  but I don't see anything.

Is there something wrong above, or am I missing something fundamental? Wemos can't do RS232? Need to enable a topic?

Many thanks for any input.

Output of `status 0`:
``
11:38:36 MQT: stat/alarm/STATUS = {"Status":{"Module":18,"FriendlyName":[],"Topic":"alarm","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
11:38:36 MQT: stat/alarm/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"External System","Uptime":"0T00:11:55","StartupUTC":"2018-04-06T10:26:41","Sleep":0,"BootCount":30,"SaveCount":60,"SaveAddress":"F6000"}}
11:38:36 MQT: stat/alarm/STATUS2 = {"StatusFWR":{"Version":"5.12.0i","BuildDateTime":"2018-03-30T12:13:02","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
11:38:36 MQT: stat/alarm/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["home-iot","indebuurt2"],"TelePeriod":300,"SetOption":["00000009","55818000"]}}
11:38:36 MQT: stat/alarm/STATUS4 = {"StatusMEM":{"ProgramSize":442,"Free":560,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
11:38:36 MQT: stat/alarm/STATUS5 = {"StatusNET":{"Hostname":"alarm-5410","IPAddress":"192.168.1.174","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"XX:C6:3A:XX:XX:XX","Webserver":2,"WifiConfig":3}}
11:38:36 MQT: stat/alarm/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A4D522","MqttUser":"user1","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:38:36 MQT: stat/alarm/STATUS7 = {"StatusTIM":{"UTC":"Fri Apr 06 10:38:36 2018","Local":"Fri Apr 06 11:38:36 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
11:38:36 MQT: stat/alarm/STATUS10 = {"StatusSNS":{"Time":"2018-04-06T11:38:36"}}
11:38:36 MQT: stat/alarm/STATUS11 = {"StatusSTS":{"Time":"2018-04-06T11:38:36","Uptime":"0T00:11:55","Vcc":3.013,"Wifi":{"AP":1,"SSId":"home-iot","RSSI":44,"APMac":"F2:9F:C2:F7:DA:EF"}}}
``

Make sure these boxes are checked before submitting your issue - Thank you!

- [x ] What hardware you are using
- [x ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

arendst, 2018-04-06:
I guess hardware serial bridge won't work quite well on a wemos as the USB port is already using it.

What I used was a software serial bridge. 
- Select two pins from the Generic Module for software serial connection on your wemos (Not Tx/Rx as they are used for the USB connection). 
- Connect your TTL to RS232 converter to it and start a serial monitoring tool like Termite.
- Set SBaudrate and Termite baudrate correctly.
- Send a MQTT message like cmnd/alarm/SSerialSend Hello there and receive it on Termite.

Once you have that functional try to get the hardware serial functional.

charlietomo, 2018-04-06:
Many thanks, but embarrassingly I am getting stuck at step 3 in terms of setting the Baudrate on the software serial bridge. Is this another elementary mistake?
``
12:10:58 CMD: SBaudrate 1
12:10:58 MQT: stat/alarm/RESULT = {"Command":"Unknown"}
``

arendst, 2018-04-06:
Command ``Baudrate`` sets the hardware serial baudrate, ``SBaudrate`` tries to set the software serial baudrate but if you didn't define software serial ports in the module definition you'll get command unknown.

charlietomo, 2018-04-07:
In `Configuration > Configure Module` I have set `SerBr TX` and `SerBr TX` to two GPIO pins. I have saved and restarted the module. What else do I need to do?

charlietomo, 2018-04-08:
Sorry to be dense but I [don't find anything in the wiki relating to termite](https://github.com/arendst/Sonoff-Tasmota/search?utf8=✓&q=termite&type=).

I have tried via MQTT and via the web console. The command `SBaudrate` gives the previous error both methods: ` MQT: stat/alarm/RESULT = {"Command":"Unknown"}`. I have tried changing the pins configured as SerBr TX and RX and get the same result.

I am either doing something dense or the software is not working as expected. Do I need to do anything else (apart from setting SerBr TX and RX to GPIO pins) in order to get the `SBaudrate` command to work?

charlietomo, 2018-06-13:
I moved to other hardware (Node-Red on Raspberry Pi) which was much easier to set up and gave me more flexibility. I hope this works out for others.

closed 2018-06-13 by charlietomo

thigger, 2023-08-05:
For anyone else (like me) who finds this issue searching for why the Software Serial commands (SBaudrate, SSerialSend) return "Unknown", it may be because you're using GPIOs between 6 and 11 (eg on an 8285 where these are free)

TasmotaSerial::isValidGPIOpin rejects these pins

` return (pin >= -1 && pin <= 5) || (pin >= 12 && pin <= 15);`

Solution is either to use a different pin (if possible), or compile Tasmota with the line changed to allow the pin you wish to use.

https://github.com/arendst/Tasmota/blob/development/lib/default/TasmotaSerial-3.6.0/src/TasmotaSerial.cpp#L126

arendst, 2023-08-05:
Any idea why Tasmota rejects these pins? Perhaps some pins are used for flash connection....

thigger, 2023-08-05:
Not on the 8285. Working fine with GPIO9 on mine now.