arendst/Tasmota #2380: Relay problem and not flashable again

xcruell, 2018-04-06:
Hello!
I have two strange problems. I flashed Sonoff-Tasmota yesterday on my d1 mini, without any problems.

no.1: When i apply power to the d1 mini with the 8ch relay connected, it just wont start. When i disconnect VCC from the 8ch relay, it startes flawlessly. no problems either when connecting after it started. 
Thats the serial output i get, when i start the d1 mini with vcc connected: 
> rd␀l��|␀�$�|␃␌␄␌�␌$�␄c|��␃�␛�{�#�␄c��o'�do'���␌#�#�lsds$�␒␛g��<�␃��n<␀␄go<��`␃

no.2:
I didnt changed the code at all, but it gives me a random error: "sonoff\sonoff_template.h:160:3: error: 'D_SENSOR_PMS5003' was not declared in this scope"
Im using Sonoff-Tasmota-5.11.1, because i get worse errors with 5.12.0.
platformio.ini content: https://pastebin.com/8VMrWP61

I hope someone can help me..

Make sure these boxes are checked before submitting your issue - Thank you!

- [x] What hardware you are using
 Wemos D1 Mini
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

Frogmore42, 2018-04-07:
How are you powering the D1 Mini?  Many computer USB ports can't supply enough power.  Try using a powered USB hub.

Frogmore42, 2018-04-07:
If you are not changing the code, you should probably use one of the pre-built binaries.

xcruell, 2018-04-07:
Im powering it over a USB port of my computer. But i dont see what difference that would make. Since it works over the USB port. It just wont start with the powered 8ch relay.

Frogmore42, 2018-04-07:
Do you have the relay connected to GPIO 0?
The relay board might be holding down the lines, which will cause the esp8266 to go into download mode.

xcruell, 2018-04-07:
Yeah thats what i thought too, but nothing is connected to D3 (GPIO 0 on d1 mini) nor D0.
It still wont boot. Only when i cut the VCC and connect it afterwards.

xcruell, 2018-04-07:
OH! I got it.. kinda.
The GPIO 0 thing got me thinking, so i tried each pin after another. And i found out that it works when nothing is connected to GPIO 0 AND GPIO 15.
But that only gives me 6 out of 8 Relays. Is there a way to workaround that?

thanks so far!

Dickesplumpel, 2018-04-07:
Look here:
https://www.forward.com.au/pfod/ESP8266/GPIOpins/index.html

xcruell, 2018-04-07:
Thank you! i just found this site seconds before i saw your comment ;)

xcruell, 2018-04-07:
Works flawlessly now.
Thank you guys very much for your help and effort!

closed 2018-04-07 by xcruell