arendst/Tasmota #2381: Timers

Lawrencezarb, 2018-04-06:
I Have a Sonoff 4CH and would like to know if it is possible to increase the number of Timers available from 16 to maybe 32?

andyd999, 2018-04-08:
The timer would need to be implemented in the managing software such as home assistant. Tasmota doesn't have any timers it's just an mqtt facilitator

curzon01, 2018-04-09:
@andyd999 - not true, since 5.12.0i: [Timers](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#timers)

Lawrencezarb, 2018-04-09:
Yes, the Timers are built into Tasmota since 5.12.0i. I just need more that 16 for the 4CH unit. The real issue is that for an on-off time setting, it requires the use of 2 of the times, which limits to 8 on-off sequences.

curzon01, 2018-04-09:
@Lawrencezarb - it's not user definable but you can try enhance MAX_TIMERS if you are familary with C code, But be aware your version then is incompatible with Thoes and resetting settings after doing that is also a good idea. If you don't understand how to handle all of that, I suggest to leave it as is.

Lawrencezarb, 2018-04-09:
I'm not familiar with C code, was hoping that Theo could make a release with the increased number of timers....

Lawrencezarb, 2018-04-09:
Well thanks curzon01, it worked, I increased the MAX_TIMERS from 16 to 32 in sonoff.h and I now have 32 timers.

closed 2018-06-02 by ascillato