arendst/Tasmota #2386: How to build sonoff-minimal

RrPt, 2018-04-07:
Hello,
my (standard) Sonoff binary is 510kB, so OTA didn't work directly. 
How can i build sonoff-minimal with my own wlan settings? (so i can directly load my code)
Or even better, how can i disable unwanted functions in the code, to get the binary smaller, so OTA works on step 1?
Haven't found that in Wiki. Is there another source of documentation?

This ist a great project!!!

Regards Rainer

RrPt, 2018-04-07:
ok, found it.
For others:
uncomment    //#define BE_MINIMAL 
in user_config.h

RrPt, 2018-04-07:
i was too fast to close.
With   #define BE_MINIMAL
i get a bin with 435kB  (the original sonoff-minimal: 422kB).
And this ist still too bit to download, but in info i see  564kB free!
The original minimal ist working

What is going wrong?

RrPt, 2018-04-07:
now my bin is 446kB, free is 556kB
and now it is working!
dont know why.

So i dont need minimal version :-)

closed 2018-04-07 by RrPt

dweston, 2018-04-18:
@RrPt Try as I may I also cannot get an OTA firmware "Upgrade by file upload" via the dialog on the Sonoff-Tasmota web interface, to work.

Steps I have taken are:

Working with the PlatformIO IDE I created a MINIMAL firmware version and flashed it via serial to a Sonoff TH16 I have apart on the bench. For good measure that firmware contains a mod to the CFG_HOLDER variable in an attempt to ensure the original locally compiled Sonoff firmware is fully overwritten.

Once installed the following two console 'status' commands result in the following output. 

`status 2` provides:
`MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.12.0l-minimal","BuildDateTime":"2018-04-18T20:21:02","Boot":4,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}`

and `status 4`:
`MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":440,"Free":560,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}`

Carving everything I can see that would materially impact the size of a non-minimal 'Sonoff' firmware version results in a 'firmware.bin' file 517056 bytes in size. According to the web server firmware upgrade tool this is still too large to upload. ???

In desperation I tried using the "Upgrade by web server" option relying on the default OTA URL http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin

This worked. And the firmware program size was slightly larger than the local firmware file I built. So I gave the "Upgrade by file upload" another go, and - Bingo - it worked.

`status 2` command produces:
`MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.12.0l","BuildDateTime":"2018-04-18T22:00:28","Boot":4,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}`
`status 4` command produces:
`MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":504,"Free":496,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}`

Not sure how these work as the firmware file size on the source MacOSX PC is still 517056 bytes in size. With this approach now apparently working hopefully it means it is possible to re-introduce some of the missing features previously carved off.