arendst/Tasmota #2387: How to config GPIO14 model as switch in user.config.h

dangyuluo, 2018-04-07:
This is not a bug report but a question. Since there's no forum I'll post it here. 

I have quite a few Sonoff Basic and I build the firmware from source code. I'd like to connect a On/Off switch to GPIO14 and GND to control the light just like the how the wall switch does. However, I cannot find a place to specify the role of GPIO14 as a switch. Does anyone know about it?

Leseratte10, 2018-04-07:
I don't think that can be configured at compile time. You need to use the Web interface - click "Configuration", then "Module" and you can set the GPIO14 to "Switch1". Then, in the Console, you can set the appropriate button mode using the "SwitchMode" command.

dangyuluo, 2018-04-07:
@Leseratte10 I'm now using the website to configure each sonoff basic. But it's a little bit trivial. So I'm looking for a way to fix it when compiling.