arendst/Tasmota #2389: Luminea ZX2820 wrong voltage and current values

RrPt, 2018-04-07:
Hello,
i am using an Luminea [ZX2820 ](http://www.luminea.info/WIFI-Smartphone-Steckdose-kompatibel-zu-Amazon-ZX-2820-919.shtml)Switch. This is identical with 'KMC 70011' and detects Power by an HLW8012.
I connected a 25W bulb and got:  P=42W, I=10.880A, U=4V
I calibrated with this 25W bulb: power to 25W, current to 109mA and Voltage to 230V.
   and got:  P=25W, I=0.109A, U=230V, all ok
Then i switched to a 60W bulb and got:  P=57W, I=0.108A, U=497V,  not ok for U and I
I connectet a heater and got:  P=1033W, I=0.105A, U=9031V,  not ok for U and I
seems like U and I are switched, may be in ZX2820 the GPIO_HLW_SEL signal is inverted.

To test this, i changed 2 lines in xdrv_03_energy.ino:

> line 204:
>     if (hlw_select_ui_flag) {
> to
>     if (!hlw_select_ui_flag) {
> line 252:
>   hlw_select_ui_flag = 0;  // Voltage;
> in
>   hlw_select_ui_flag = 1;  // Voltage;

I connected a 25W bulb and got:  P=41W, I=0.148A, U=294V
I calibrated with this 25W bulb: power to 25W, current to 109mA and Voltage to 230V.
   and got:  P=25W, I=0.109A, U=230V, all ok
Then i switched to a 60W bulb and got:  P=57W, I=0.237A, U=229V,  ok
I connectet a     heater and got:  P=1064W, I=4.377A, U=223V,  ok
I connectet a 2. heater and got:  P=2035W, I=8.452A, U=218V,  ok

So it ist true, in my module the GPIO_HLW_SEL ist inverted compared to 'KMC 70011'

How can the code be changed (in GitHub), so that both modules are working. 
Is ist possible to add a GPIO_HLW_SEL_INV?

By the Way:
I think the CalibrationValue is the reference for a device with a power of 1000W.
The CalibrationValue of the voltage then shoud be 230V (not 220V). This had changed in europe since th year 1987.
The calibrationValue for the current then should be 4.348A (= 1000/230) instead of the actual 4.545A

Rainer

RrPt, 2018-06-18:
Hi,
is there anybody who can ad this two lines of Code?
Or what is the correct way to push this Issue?
Rainer

closed 2018-08-17 by stale[bot]

Guzzi-Charlie, 2018-12-05:
Hello Dear all,
I also have some issues with a Luminea WIFI-Plug. In my case it is a SF-500 (ZX-2820-675). IO got it flashed without problems and switching via Web-IF or manual works fine.

But I have the same problem as RrPt who started this thread. I'm receiving more or less no correct values from the power measurement.

e.g. by connecting a 60 bulb
[image]

Is there any possibility for calibration?