arendst/Tasmota #2390: Sonoff TH dry contacrs

paveleremin, 2018-04-07:
Any body know what should be done to get dry contacts instead of main voltage at Sonoff TH?

Stevehans, 2018-04-07:
I butchered a basic to do this, just cut the tracks, you have to cut both sides of the board, and also check continuity afterwards, you don't want to frazzle whatever is on the dry contacts!

paveleremin, 2018-04-07:
@Stevehans could you provide some image please?

Stevehans, 2018-04-07:
2nd hit on google:-
https://www.youtube.com/watch?v=FdfMTlo8hFM

paveleremin, 2018-04-08:
@Stevehans it's SV but my question about TH

paveleremin, 2018-04-08:
I found only this information
[image]
Here is my implementation:
[image]
but seems still mains at the output

kainron, 2019-01-14:
Cut input just before the relay attach the relay side if the input you just cut to the E terminal.
Ground and output will now be dry

Stevehans, 2019-01-14:
Check with an ohm meter, think I had to cut both sides of the board.

Regards
Steve

lancelon, 2019-05-10:
Anyone know if this can be done with a POW?