arendst/Tasmota #2392: Mysterious "CMD M105" being issued at interval

Mobyn, 2018-04-08:
NodeMCU running 5.12.0 as 18 Generic flashed via esptool
Relay 1 on GPIO 12
Relay 2 on GPIO 15

I'm getting a ghost command pop up on the console every 2 seconds

13:10:28 CMD: 
13:10:28 MQT: stat/Garage_Sonoff/RESULT = {"Command":"Unknown"}
13:10:28 CMD: M105
13:10:28 MQT: stat/Garage_Sonoff/RESULT = {"Command":"Unknown"}
13:10:30 CMD: 
13:10:30 MQT: stat/Garage_Sonoff/RESULT = {"Command":"Unknown"}
13:10:30 CMD: M105
13:10:30 MQT: stat/Garage_Sonoff/RESULT = {"Command":"Unknown"}

Not actually creating problems except when I'm trying use the console effectively. 

Amazing piece of code, thank you so much for building and supporting it!

Jason2866, 2018-04-08:
Which Home Automation system do you use?

Mobyn, 2018-04-09:
HASS. I fixed the issue by flashing again. It was obviously something I did w/ console commands

Geitde, 2018-04-11:
M105 sounds like the temperature polling from a 3D Printer.

Mobyn, 2018-04-11:
You are TOTALLY right! Odd that it's coming through the console of sonoff webserver. My other mqtt messages don't