arendst/Tasmota #2394: remote configuration backup

or78rXxZ, 2018-04-08:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

This is a generic request for Tasmota. Is it possible for a server to pull the configuration from a device that has Tasmota installed? I have not been able to find this. If not could this enhancement be added?

curzon01, 2018-04-08:
Possible ideas:
# use saved (Backup Config) configuration file and post it using curl
# if already ble to use MQTT wirh your device use MQTT command Backlog and set each neccessary setting command by command

your question is too common, there are a lot of possibilities depending on condition situation you have

curzon01, 2018-04-08:
curl and POST request to Url for Web GUI “Backup Configuratuon“

or78rXxZ, 2018-04-08:
wget sonoff_ip_address/dl
This should be documented
Thanks

closed 2018-04-08 by or78rXxZ

arendst, 2018-04-10:
Nice to know ;-)