arendst/Tasmota #2395: ds18b20 temp difference between devices

andyd999, 2018-04-08:
I have 2x ds18b20 temp probes on test. 1x powered by sonoff th10 plugged in direct, the other probe wired onto the gpio of a wemos d1 mini with 4k7 pullup ressistor.  they are reading about 0.5 celcius difference.  I ahev tried 3.3 and 5v on the wemos without any luck.  2 questions? 1) what is best voltage on the wemos 3.3 or 5v? and 2) is the difference just different proves or different hardware controls? i cant easily swap the probes around

Frogmore42, 2018-04-09:
All esp8266 based devices are normally rated for 3V3 max.  While they might work with 5V on the input for some period of time, that exceeds the absolute maximum input voltage.  With a 4k7 to 5V, they would likely last longer, but it is not recommended.

0.5C is within the specification of accuracy for the DS18B20:
 ±0.5°C Accuracy from -10°C to +85°C 
https://datasheets.maximintegrated.com/en/ds/DS18B20.pdf

Also, that is also possible due to environmental differences, i.e. unless they are in a very accurately temperature controlled medium, they will be different.

andyd999, 2018-04-09:
Thanks for explaining

closed 2018-04-09 by andyd999