arendst/Tasmota #2396: project name

andyd999, 2018-04-08:
since uploading tasmota i havent changed the default project name in the files.. after each device has been setup i`ve manually changed the mqtt topic etc.  will this cause issues when doing ota updates? is there a way of updating the project names on already flashed devices?

curzon01, 2018-04-10:
Saving flash live time and your time - simply use Web GUI "Reset Config" or command `Reset 1` - it results in same as changing cfg holder.

@andyd999 - why not simply change config values using the powerful [Commands](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands) - you don't need to change the project name within src if you simpy can play with `FullTopic` and set this like you want, e .g. `FullTopic myhome/%prefix%/tasmota/%topic%/` and `Topic Light1` will result in topics like `myhome/stat/tasmota/Light1/...`

andyd999, 2018-04-10:
Thanks guys. Got most of that's but somebody above my head. So going forward I should keep each project name the same "sonoff" for example. Can i use commands to change the project name for devices already flashed ?

curzon01, 2018-04-10:
The project name you mentioned is used for define the default MQTT topic, so yes, you can change it for already flashed devices, use e. g. the Web GUI "Console" and the command

`Topic`

to show the current name and

`Topic newprojectname`

to change it. It's already described in the [Command Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)

dastrix80, 2018-05-06:
Hi All,

Further to the project name, what significant does this have?

I have two units configured identically (or at least I think i do). Same firmware, same board - Sonoff basic.

The Garden device does not receive the OH2 box sends, despite it appearing OK in the log. The AV Device works perfectly.

**Garden Device:**

00:00:00 Project Unused Garden (Topic gardenlights, Fallback gardenlights, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Connecting to AP1 Tard in mode 11N as gardenlights-3744...

**AV Device:**

00:00:00 Project sonoff Garage AV (Topic garage, Fallback garage, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Connecting to AP1 Tard in mode 11N as garage-3651...

```
19:27:47.613 [INFO ] [smarthome.event.ItemCommandEvent     ] - Item 'GarageAVReceiver' received command ON
19:27:47.615 [INFO ] [smarthome.event.ItemStateChangedEvent] - GarageAVReceiver changed from NULL to ON
19:27:52.405 [INFO ] [smarthome.event.ItemCommandEvent     ] - Item 'GarageAVReceiver' received command OFF
19:27:52.407 [INFO ] [smarthome.event.ItemStateChangedEvent] - GarageAVReceiver changed from ON to OFF
19:27:53.808 [INFO ] [smarthome.event.ItemCommandEvent     ] - Item 'GardenLights' received command ON
19:27:53.808 [INFO ] [smarthome.event.ItemStateChangedEvent] - GardenLights changed from OFF to ON
19:27:55.164 [INFO ] [smarthome.event.ItemCommandEvent     ] - Item 'GardenLights' received command OFF
19:27:55.166 [INFO ] [smarthome.event.ItemStateChangedEvent] - GardenLights changed from ON to OFF

```

arendst, 2018-05-06:
Project name is only used as a placeholder for default topic name and (in the past) for Arduino IDE binary filename as the IDE failed to use a filename I wanted.

Today just leave the project name as is and as @Frogmore42 explained above change topic at will using commands.

dastrix80, 2018-05-07:
Thank you for the explaination arendst

closed 2018-06-02 by ascillato