arendst/Tasmota #2399: Timers/Timer commands

meirlebel, 2018-04-09:
Hi,

I have installed the last Tasmota release version on a NodeMcu module and chose "Sonoff 4CH Module".
All looks working as expected, but The Timers/Timer commands.
When trying to use get the message: "stat/sonoff/RESULT = {"Command":"Unknown"}".

Below is the output of "Status 0" command.

Please advice,
Tn'x in advance,
Meir

07:25:25 RSL: stat/sonoff/STATUS = {"Status":{"Module":7,"FriendlyName":"Sonoff","Topic":"sonoff","ButtonTopic":"0","Power":4,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
07:25:25 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T05:57:21","Sleep":0,"BootCount":30,"SaveCount":183,"SaveAddress":"F6000"}}
07:25:25 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
07:25:25 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"ONOFF","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000009"}}
07:25:25 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":22,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3}}
07:25:25 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3066","IPAddress":"10.100.102.6","Gateway":"10.100.102.1","Subnetmask":"255.255.255.0","DNSServer":"10.100.102.1","Mac":"18:FE:34:FC:AB:FA","Webserver":2,"WifiConfig":1}}
07:25:25 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.dioty.co","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FCABFA","MqttUser":"meirlebel@gmail.com","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
07:25:25 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Apr 09 05:25:25 2018","Local":"Mon Apr 09 07:25:25 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2}}
07:25:25 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-04-09T07:25:25"}}
07:25:25 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-04-09T07:25:25","Uptime":"0T05:57:21","Vcc":2.736,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","POWER4":"OFF","Wifi":{"AP":1,"SSId":"ONOFF","RSSI":100,"APMac":"E8:74:E6:FE:D7:A1"}}}

curzon01, 2018-04-09:
Timers are implemented since 5.12.0i - see [_releasenotes.ino](https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/_releasenotes.ino):
> * 5.12.0i
> * Add 16 timers using commands Timer and Timers (#1091)

You are using 5.12.0
> RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.12.0"

meirlebel, 2018-04-09:
Tn'x for the quick response.

Where can i find the 5.12.0i or 5.12.0j bin file?
I kind a like the idea of using already compiled and not building a development environment.

Tn'x
Meir

reloxx13, 2018-04-09:
if you dont want to compile you need to wait for the next release.

closed 2018-06-02 by ascillato