arendst/Tasmota #2400: 12kB more code with 5.12.0j

0n3-70uch, 2018-04-09:
Hello, I compiled the version 5.12.0i and 5.12.0j. But the version 5.12.0j use 12kB more of flash. I excluded the timer option with these options:

```
#ifdef USE_TIMERS
#undef USE_TIMERS
#endif

	#ifdef USE_TIMERS_WEB
	#undef USE_TIMERS_WEB
	#endif

	#ifdef USE_SUNRISE
	#undef USE_SUNRISE
	#endif
```

- [X] What hardware you are using: Sonoff Basic
- [X] Provide the output of command ``status 0``: not necessary
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

Jason2866, 2018-04-09:
Is user_config_override really used? You have to enable this option in your IDE!
In ATOM.io you have to see a yellow warning.

0n3-70uch, 2018-04-09:
Yes it's enabled. With active `USE_TIMERS` I get round about +30kB of code.

arendst, 2018-04-10:
That's what the user_config_override.h file is for.

I enable most new features in the development branch to get people testing the new functionality. What you do not need/want just disable it as you perfectly did in the first entry. Although disabling define USE_TIMERS only is enough to get all Timer code vanished from the distribution.

The master branch will have an ever bigger exclusion config but I probably won't be able to keep it below the magic OTA high water mark...

0n3-70uch, 2018-04-10:
Hi Theo,

thanks for your help. At the moment it's necessary to exclude the `USE_TIMERS_WEB` and `USE_SUNRISE` . Here you can check the file size (with my user config):

Timers active = 530.0kB

`#undef USE_TIMERS` = Compiler error 

> .pioenvs/sonoff/src/sonoff.ino.cpp.o:(.text._Z13PublishStatush+0x68): undefined reference to `GetSun(unsigned char)'
> .pioenvs/sonoff/src/sonoff.ino.cpp.o: In function `PublishStatus(unsigned char)':
> sonoff.ino.cpp:(.text._Z13PublishStatush+0x4d6): undefined reference to `GetSun(unsigned char)'
> sonoff.ino.cpp:(.text._Z13PublishStatush+0x4e9): undefined reference to `GetSun(unsigned char)'
> collect2: error: ld returned 1 exit status
> *** [.pioenvs/sonoff/firmware.elf] Error 1

`#undef USE_TIMERS` and `USE_TIMERS_WEB` = same compiler error

`#undef USE_TIMERS` and `USE_TIMERS_WEB` and `USE_SUNRISE` = 512.5kB

Version 5.12.0i without `USE_SUNRISE` = 499.7kB

So something is wrong :-).

arendst, 2018-04-10:
Ah verified. Will investigate.

arendst, 2018-04-10:
Found it. In file sonoff.ino change line 1309 from
```
#ifdef USE_SUNRISE
```
in to
```
#if defined(USE_TIMERS) && defined(USE_SUNRISE)
```

arendst, 2018-04-10:
#1592 will be solved soon.

When timers are disabled all code should be disabled. Pls provide user_config info for deciding where the 12.9 kB could come from.

arendst, 2018-04-10:
Are you sure you used the same esp/arduino library while comparing 5.12.0i with 5.12.0j?

0n3-70uch, 2018-04-10:
Yes now I'm 100% sure. I downloaded version 5.12.0k (thanks for the quick fix) and the old 5.12.0i:
https://github.com/arendst/Sonoff-Tasmota/commit/d564615b7d70b493925f2045699ed94b36e7ccdd

The difference is still the same. Crazy.

0n3-70uch, 2018-05-05:
I tested many different compiler options. I'm not sure but I think the difference comes from different ESP libraries (2.3.0 / 2.4.0 / 2.4.1).

Greetings Fabi

closed 2018-05-05 by 0n3-70uch