arendst/Tasmota #2401: Each button to mqtt but not linked to relays

antibill51, 2018-04-09:
Hello. I am looking to use a "sonoff 4ch tasmota" connected to home assistant. I would like to retrieve the information in mqtt when pressing a button is done, and that for each button, and that behind that, home assistant decides whether or not we change the state of the relay associated. I get there with buttontopic but only for the first button, the others remain linked to power2 power3 and power4. An idea of ​​how to proceed?

Thank you in advance.

nosaj66au, 2018-04-12:
I agree, I have the same feature request #2360 but I need to use it on a Generic device. From memory I have seen this request before that but could not find the reference.

What do you then @arendst ?

nosaj66au, 2018-04-12:
Ok found some simlar requests but not the one I was thinking of.
#2281 and #2020

antibill51, 2018-04-14:
The only solution now is it to use easyesp?

arendst, 2018-04-14:
Or upgrade to 5.12.0l and use a rule for every button. See https://github.com/arendst/Sonoff-Tasmota/wiki/Rules

For a Sonoff 4CH possible rules could be:

``on button1#state do publish cmnd/button1/power %value% endon on button2#state do publish cmnd/button2/power %value% endon on button3#state do publish cmnd/button3/power %value% endon on button4#state do publish cmnd/button4/power %value% endon``

Enjoy.

antibill51, 2018-04-14:
Perfect job !
And how to remove automatic link between button and relay, but keep relay command with mqtt?

arendst, 2018-04-14:
With rules active there is no relation between button and relay as that is what you want using the rules. 

The relay itself are still accessible with mqtt like cmnd/4ch/power1 on

antibill51, 2018-04-14:
Thanks ! I will try soon.