arendst/Tasmota #2406: support for raw irsend

vandrham, 2018-04-09:
I would like to have an additional irremote type, e.g. USE_IR_RAW
This type would allow a raw ir command to be sent given a specific byte array and frequency.

I currently have my own sketch running on an ESP32 that allows me to remotely control a pellet stove, I would love to use the Tasmota firmware running on my sonoff to control it instead.

dasMopo, 2018-04-10:
Actually there is already support for raw IR codes, cf. IRsend command here:

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

If that doesn't suit your needs or you prefer to define custom commands instead of sending raw IR codes over mqtt [smth I plan to do for controlling my custom build dimmable LED ceiling lamps], take a look at how support for those hvac devices was done.

From what I've seen, adding some custom commands that issue a number of corresponding IR codes should be a rather simple task, thanks to Theo's nice dynamic function calling and command mailbox usage.

Go see for yourself ;)

melogonzaga, 2018-08-28:
I Didn't find how to send the raw data. Can you give me an example?