arendst/Tasmota #2409: Cannot automatically configure unique MQTT topic

ninjalf2, 2018-04-10:
Is it possible to configure something in the code that will always result in a unique MQTT topic without having to go and configure it manually? I could not find anything about such a feature. I only found something called the fallback topic which seems is not supposed to be used intentionally. However, this is exactly what I am looking for.

Any help is appreciated
Thanks in advance

Leseratte10, 2018-04-10:
You can use the placeholders %d, %04d or %X in the MQTT topic or the MQTT client name. They are replaced with the decimal chip ID, the short decimal chip ID and the hexadecimal chip ID.

ninjalf2, 2018-04-10:
I have tried this but the topic just looks like this:

```
12:49:34 MQT: stat/%X/RESULT = {"POWER":"OFF"}

12:49:34 MQT: stat/%X/POWER = OFF
```

Leseratte10, 2018-04-10:
This feature was recently added, in 5.12.0b. If you are running a previous version, you need to update. If you already run 5.12.0b or newer and %X still doesn't work, can you please try %d instead?

ninjalf2, 2018-04-10:
%d also doesn't work. I will try updating to 5.12.0b

ninjalf2, 2018-04-10:
Updated to development branch (5.12.0k) and %d, %04d, %06d, and %06X work but %X does not. Are %X and %06X supposed to be the same? Because in that case I will just use %06X

```
00:00:06 MQT: stat/Sonoff_%X/RESULT = {"POWER":"OFF"}

00:00:06 MQT: stat/Sonoff_%X/POWER = OFF
```

Leseratte10, 2018-04-10:
%X seems not to be working in the current implementation. If %06X works for you, just use that. Or use %d, if you don't mind the ID being decimal.

ninjalf2, 2018-04-10:
I am using %06X as I prefer hex. Should I close this or should it remain open until %X is fixed?

arendst, 2018-04-10:
Close it as %X won't be fixed.

closed 2018-04-10 by arendst