arendst/Tasmota #2411: Why status is not retained?

demian85, 2018-04-10:
- [x] Sonoff Basic
- [x] Provide the output of command ``status 0``

```
{"Status":{"Module":1,"FriendlyName":"Desk lamp","Topic":"sonoff-desk","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
{"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"1T19:13:17","Sleep":0,"BootCount":9,"SaveCount":51,"SaveAddress":"F9000"}}
{"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-04-07T14:37:16","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
{"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"Garchatel WiFi","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000009"}}
{"StatusMEM":{"ProgramSize":493,"Free":508,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
{"StatusNET":{"Hostname":"sonoff-desk-7625","IPAddress":"192.168.0.12","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"200.42.4.198","Mac":"5C:CF:7F:72:DD:C9","Webserver":2,"WifiConfig":3}}
{"StatusMQT":{"MqttHost":"m10.cloudmqtt.com","MqttPort":17279,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_72DDC9","MqttUser":"saeyhxpw","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
{"StatusTIM":{"UTC":"Tue Apr 10 14:57:40 2018","Local":"Tue Apr 10 15:57:40 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
{"StatusSNS":{"Time":"2018-04-10T15:57:40"}}
{"StatusSTS":{"Time":"2018-04-10T15:57:40","Uptime":"1T19:13:17","Vcc":3.468,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Garchatel WiFi","RSSI":58,"APMac":"B8:EE:0E:42:3A:36"}}}```

I'm struggling to understand why when subscribing to the channel `stat/sonoff-desk/POWER` I get nothing. How can I know the state of the device when connecting for the first time if the status is not retained? Is it retained? Am i missing something?
For example, when using MQTT dashboard, the initial status is unknown (n/a) in the subscribe tab.
When using MQTT dash, the switch is not always shown in the current state because the app does not know the device status.
Which topic should I use to find out the current state of the device?

Thanks.

curzon01, 2018-04-10:
see [Wiki Commands - Main](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#main)

`PowerRetain ON`

x0010, 2018-04-11:
https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Overview

You need to publish a command to be able to get something back from your subscribed topic. 
To get something on your `stat/sonoff-desk/POWER` you first need to send a `cmnd/sonoff-desk/POWER` (without any value after POWER so you will not set state) so that your mqtt-broker will respond.
If you 
OR, you just need to change your teleperiod to a few seconds (which can be bad if you have many devices but I think 10 seconds is the lowest) and instead of subscribing to `stat/sonoff-desk/POWER` you subscribe to `tele/sonoff-desk/STATE`.

I hope that clears it out.

demian85, 2018-04-11:
PowerRetain ON worked. Thanks!
However, I think it should be the default value, as anyone would expect to
connect and get the current device status. Am I wrong?

curzon01, 2018-04-12:
Retain should be the exception, not the default because retain messages do not represent the reality and this can cause a lot of issues when you are not aware about that (example are a increase number of issues e.g. with Hass which uses retain by default).
The situation between a true device status and that what MQTT broker reflacts your environment can be asynchronous. This might be ok for e.g. measured values like temperature where you want to know the last known value but as already explained, this causes shake of the head for users who are not aware about retain, it's meaning and interpretation.

closed 2018-06-04 by ascillato

jeroenst, 2021-11-15:
Configurable retain for status messages would be a great option. Now I have to poll each 5 minutes and send retained messages to another statustopic to have them available for other non persistent clients.

curzon01, 2021-11-15:
[SensorRetain](https://tasmota.github.io/docs/Commands/#sensorretain)

jeroenst, 2021-11-15:
> [SensorRetain](https://tasmota.github.io/docs/Commands/#sensorretain)

I don't need sensor retain but statusretain

curzon01, 2021-11-15:
statusretain for what [status](https://tasmota.github.io/docs/Commands/#status)?

jeroenst, 2021-11-15:
For all status messages together

ascillato, 2021-11-15:
> Configurable retain for status messages would be a great option. Now I have to poll each 5 minutes and send retained messages to another statustopic to have them available for other non persistent clients.

The discussion of this is at https://github.com/arendst/Tasmota/pull/11109

curzon01, 2021-11-15:
don't think such a PR will merged