arendst/Tasmota #2415: TSL2561 not working

Jason2866, 2018-04-10:
Make sure these boxes are checked before submitting your issue - Thank you!

- [x ] What hardware you are using
NodeMCU
- [x ] Provide the output of command ``status 0``
21:07:45 CMD: Status 0
21:07:45 RSL: Group 0, Index 1, Command STATUS, Data 0
21:07:45 MQT: stat/sonoff-5D3631/STATUS = {"Status":{"Module":18,"FriendlyName":[],"Topic":"sonoff-5D3631","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
21:07:45 MQT: stat/sonoff-5D3631/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:10","StartupUTC":"2018-04-10T19:03:35","Sleep":1,"BootCount":94,"SaveCount":129,"SaveAddress":"FA000"}}
21:07:45 MQT: stat/sonoff-5D3631/STATUS2 = {"StatusFWR":{"Version":"5.12.0k","BuildDateTime":"2018-04-10T20:20:43","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:07:45 MQT: stat/sonoff-5D3631/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"SetOption":["00000009","55818000"]}}
21:07:45 MQT: stat/sonoff-5D3631/STATUS4 = {"StatusMEM":{"ProgramSize":497,"Free":504,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
21:07:45 MQT: stat/sonoff-5D3631/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5D3631-5681","IPAddress":"192.168.2.122","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"60:01:94:5D:36:31","Webserver":2,"WifiConfig":3}}
21:07:45 MQT: stat/sonoff-5D3631/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"sonoff-5D3631","MqttClient":"sonoff-5D3631","MqttUser":"openhabian","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:07:45 MQT: stat/sonoff-5D3631/STATUS7 = {"StatusTIM":{"UTC":"Tue Apr 10 19:07:45 2018","Local":"Tue Apr 10 21:07:45 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"06:32","Sunset":"19:56"}}
21:07:45 MQT: stat/sonoff-5D3631/STATUS10 = {"StatusSNS":{"Time":"2018-04-10T21:07:45"}}
21:07:45 MQT: stat/sonoff-5D3631/STATUS11 = {"StatusSTS":{"Time":"2018-04-10T21:07:45","Uptime":"0T00:04:10","Vcc":2.871,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}}

WebConsole Log (DEBUG Mode) after Start:
00:00:00 CFG: Loaded from flash at FB, Count 128
00:00:00 APP: Boot Count 94
00:00:00 Project sonoff NodeMCU (Topic sonoff-5D3631, Fallback sonoff-5D3631, GroupTopic sonoffs) Version 5.12.0k-2_3_0
00:00:00 CFG: Saved to flash at FA, Count 129, Bytes 2048
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 Jason_Home_WLAN in mode 11N as sonoff-5D3631-5681...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on sonoff-5D3631-5681 with IP address 192.168.2.122
00:00:07 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: tele/sonoff-5D3631/LWT = Online (retained)
00:00:07 MQT: cmnd/sonoff-5D3631/POWER = 
00:00:07 MQT: Subscribe to cmnd/sonoff-5D3631/#
00:00:07 MQT: Subscribe to cmnd/sonoffs/#
00:00:07 MQT: Subscribe to cmnd/sonoff-5D3631/#
00:00:07 MQT: tele/sonoff-5D3631/INFO1 = {"Module":"Generic","Version":"5.12.0k","FallbackTopic":"sonoff-5D3631","GroupTopic":"sonoffs"}
00:00:07 MQT: tele/sonoff-5D3631/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5D3631-5681","IPAddress":"192.168.2.122"}
00:00:07 MQT: tele/sonoff-5D3631/INFO3 = {"RestartReason":"Software/System restart"}
00:00:09 APP: (UTC) Tue Apr 10 19:03:44 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
21:03:45 HTP: Main Menu
21:03:47 HTP: Console
21:03:49 I2C: TSL2561 found at 0x39
21:03:50 MQT: tele/sonoff-5D3631/STATE = {"Time":"2018-04-10T21:03:50","Uptime":"0T00:00:15","Vcc":2.937,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}

Any idea? I enabled the sensor in user comfig.h

Jason2866, 2018-04-11:
Yes, correct PIN. Device is recognized; see timestamp 21:03:49 I2C: TSL2561 found at 0x39
but not working :-(
I use several sensors (DHT22, BMP280, BME280, IR send and receiver, 433Mhz receiver on SerialBridge) the all work...

renne, 2018-04-11:
Same problem here with a GY-2561 breakout board and 5.12.0i. I2CScan detects a device at 0x39, but TSL2561 doesn't work.
`18:14:51 RSL: stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x39 0x48 0x76"}`

ADS1117 (0x48) and BME680 (0x76) work fine. I also triple-checked `#define USE_TSL2561` is uncommented before compiling.

RaymondMouthaan, 2018-04-12:
Just re-flashed my Wemos with 5.12.0k with TSL2561 enabled using user_config_override.h ...

```
#ifdef USE_I2C
#undef USE_TSL2561
#define USE_TSL2561                            // Add I2C code for TSL2561 sensor using library Joba_Tsl2561 (+2k3 code)
#endif
``` 
The TSL is detected (takes a couple of seconds) after a reboot.

```
13:27:11 MQT: tele/wemos-5/SENSOR = {"Time":"2018-04-12T13:27:11","TSL2561":{"Illuminance":246.072}}
``` 

My config is the same as described on the [Wemos-D1-Mini](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini#tsl2561-luminosity-sensor-i2c).

Jason2866, 2018-04-12:
I tried my TSL2561 now on a Wemos D1 mini. Same Firmware flashed (5.12.0k). Here it works. Strange
On the device before (NodeMCU) it didnt. Maybe the connected AM2301 on D5 is disturbing in some way.
On both i tried the Tsl2561 on the same pins (D3 and D4)

renne, 2018-04-16:
Just tried 5.12.0l on [ESP8266 SMD Adapter Breakout Board R2.1](http://www.electrodragon.com/product/esp8266-smd-adapter-board/), generic module, SDA on GPIO12 and SCL on GPIO14. Again a device is detected at 0x39 by I2CScan but no sensor data.

joba-1, 2018-05-11:
Hi,
I had the same issue (and some more), so I wrote a new TSL2561 lib for that. Theo has included it in his firmware. It was around 5.12.* as an alternative lib with another #define to enable it, and later as a replacement. It works fine for me in V5.13.1a showing I should go enjoy the evening sun at 2607.293 lx :)

renne, 2018-05-12:
Problem still exists with Tasmota Version 5.13.1a. Is 0x39 the correct I²C-address for your library?

Jason2866, 2018-05-12:
It works i have 2 sensors connected (BMP280 and TLS)
[image]

Jason2866, 2018-05-12:
@renne Try a different device. I tested the sensor again on my NodeMCU.
Guess what? It doesnt work here. Every other sensor does work with the NodeMCU.
So i think the TSL sensor is a "diva"

joba-1, 2018-05-12:
@renne 0x39 is one of the three possible adresses for tsl2561 my lib checks.

But if it cannot find a device on one of the adresses it just does nothing. Especially no message containing 0x39. Where did you get that adress from?

joba-1, 2018-05-12:
ah, and maybe try to use pullup resistors for the i2c bus.

joba-1, 2018-05-13:
looks all fine to me. Can you increase the debug level and post the
complete log from boot to, say, 2nd tele message?
Maybe there is something in it that tells me or Theo something.

Or try the testing.ino example I provide with the lib, and if this doesn’t
work, provide the log.
-- 
iphone so i typo

renne, 2018-05-13:
Testing.ino uses `Wire.begin();` Does this mean SCL=GPIO5 and SDA=GPIO4?

joba-1, 2018-05-13:
yes, aka D1 for SCL and D2 on Node MCUs.
-- 
iphone so i typo

joba-1, 2018-05-13:
yes, fine, wiring and chip work.

And yes, fluctuations of raw readings are normal. You can better see them
in the Utility.ino example. Generally accuracy is best with exposure and
gain set, so that raw sensor values are somewhere in the middle of the
range. That is what my autogain function aims for and what tasmota uses.

-- 
iphone so i typo

renne, 2018-05-13:
After running the Testing.ino the TSL2561 worked once with Tasmota. When I held my hand over the TSL2561 to check for value changes it vanished. This is LOG_LEVEL_DEBUG_MORE:

```
00:00:00 CFG: aus Flash geladen am FB, zählen 9
00:00:00 APP: Anzahl Startvorgänge 5
00:00:00 Projekt sonoff Sonoff (Topic sonoff, Fallback DVES_0BF89C, GroupTopic sonoffs) Version 5.13.1a-2_4_1
00:00:00 CFG: in Flash gespeichert am FA, zählen 10, Bytes 2560
00:00:00 WIF: Prüfe Verbindung...
00:00:00 WIF: Verbindungsversuch...
00:00:00 WIF: verbinden mit AP1 BartschNet in Modus 11N wie sonoff-6300...
00:00:01 WIF: Prüfe Verbindung...
00:00:01 WIF: Verbindungsversuch...
00:00:02 WIF: Prüfe Verbindung...
00:00:02 WIF: Verbindungsversuch...
00:00:04 WIF: Prüfe Verbindung...
00:00:04 WIF: Verbindungsversuch...
00:00:05 WIF: Prüfe Verbindung...
00:00:05 WIF: Verbindungsversuch...
00:00:05 APP: (UTC) Sun May 13 16:27:54 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
17:27:55 WIF: Prüfe Verbindung...
17:27:55 WIF: verbunden
17:27:55 DNS: initialisiert
17:27:55 HTP: Web-Server aktiv bei sonoff-6300.local mit IP-Adresse 192.168.100.42
17:27:56 MQT: Verbindungsversuch...
17:27:57 DNS: Suchanfrage abgeschlossen. MQTT-Services gefunden 0
17:27:57 MQT: Verbindung fehlgeschlagen aufgrund von :1883, rc -2. Wiederversuch in 10 sek
17:28:08 MQT: Verbindungsversuch...
17:28:09 DNS: Suchanfrage abgeschlossen. MQTT-Services gefunden 0
17:28:09 MQT: Verbindung fehlgeschlagen aufgrund von :1883, rc -2. Wiederversuch in 10 sek
17:28:12 HTP: Konsole
17:28:17 WIF: Prüfe Verbindung...
17:28:17 WIF: verbunden
17:28:20 MQT: Verbindungsversuch...
17:28:21 DNS: Suchanfrage abgeschlossen. MQTT-Services gefunden 0
17:28:21 MQT: Verbindung fehlgeschlagen aufgrund von :1883, rc -2. Wiederversuch in 10 sek
17:28:21 SHT: Sensor hat ACK-Befehl nicht ausgeführt
17:28:21 I2C: TSL2561 gefunden bei 0x39
17:28:22 RSL: tele/sonoff/STATE = {"Time":"2018-05-13T17:28:22","Uptime":"0T00:00:33","Vcc":3.013,"Wifi":{"AP":1,"SSId":"BartschNet","RSSI":100,"APMac":"34:81:C4:2A:17:F3"}}
17:28:33 MQT: Verbindungsversuch...
17:28:34 DNS: Suchanfrage abgeschlossen. MQTT-Services gefunden 0
17:28:34 MQT: Verbindung fehlgeschlagen aufgrund von :1883, rc -2. Wiederversuch in 10 sek
17:28:39 WIF: Prüfe Verbindung...
17:28:39 WIF: verbunden
17:28:45 MQT: Verbindungsversuch...
17:28:46 DNS: Suchanfrage abgeschlossen. MQTT-Services gefunden 0
17:28:46 MQT: Verbindung fehlgeschlagen aufgrund von :1883, rc -2. Wiederversuch in 10 sek
17:28:54 SHT: Sensor hat ACK-Befehl nicht ausgeführt
17:28:55 RSL: tele/sonoff/STATE = {"Time":"2018-05-13T17:28:55","Uptime":"0T00:01:06","Vcc":3.013,"Wifi":{"AP":1,"SSId":"BartschNet","RSSI":96,"APMac":"34:81:C4:2A:17:F3"}}
17:28:57 MQT: Verbindungsversuch...
17:28:58 DNS: Suchanfrage abgeschlossen. MQTT-Services gefunden 0
17:28:58 MQT: Verbindung fehlgeschlagen aufgrund von :1883, rc -2. Wiederversuch in 10 sek
17:29:02 WIF: Prüfe Verbindung...
17:29:02 WIF: verbunden
17:29:09 MQT: Verbindungsversuch...
17:29:10 DNS: Suchanfrage abgeschlossen. MQTT-Services gefunden 0
17:29:10 MQT: Verbindung fehlgeschlagen aufgrund von :1883, rc -2. Wiederversuch in 10 sek
```
I do not have SHT sensors and the TSL2561 was the only device connected via I2C.

joba-1, 2018-05-13:
Hm, so sollte das aussehen:

17:49:19 I2C: TSL2561 gefunden bei 0x39
17:49:20 HTP: Hauptmenü
17:49:20 Beleuchtungsintensität g:1, e:1, f:5780, i:2448 -> 68.618 lx
17:49:20 MQT: tele/lowvolt2/STATE =
{"Time":"2018-05-13T17:49:20","Uptime":"0T00:00:13","Vcc":3.366,"POWER1":"OFF","POWER2":"OFF","Dimmer":24,"Color":"1C3D04","HSBColor":"95,93,24","Channel":[11,24,2],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"job.fritz.ssid","RSSI":60,"APMac":"00:18:39:C6:B6:32"}}
17:49:20 Beleuchtungsintensität g:1, e:1, f:5780, i:2448 -> 68.618 lx
17:49:20 MQT: domoticz/in =
{"idx":13,"nvalue":0,"svalue":"69","Battery":76,"RSSI":6}
17:49:20 MQT: tele/lowvolt2/SENSOR =
{"Time":"2018-05-13T17:49:20","TSL2561":{"Illuminance":68.618}}

Leider hab ich keine Ahnung was da schief läuft. Den einzigen Unterschied,
den ich sehe, ist dass mein MQT läuft und Domoticz konfiguriert ist.
Auf der Haupt-Webseite deines Devices wird nichts angezeigt?

-- 
iphone so i typo

Jason2866, 2018-05-13:
Do you have your ESP device on a breadboard?
I have problems if i connect I2C devices via the breadboard.
I have them connected via dupont jumper wire cables -> no problems (even when i connect more sensors)

Jason2866, 2018-05-13:
Is your mqtt broker off?

renne, 2018-05-13:
@joba-1
Testing.ino works fine. Web-console and serial logging of Tasmota show the output from my comment above, no sensor data. Once the sensor showed up on the main page, directly after running Testing.ino. When I waved my hand 10 cm above the TSL2561 to check the values it vanished. I couldn't reproduce it. If I flash Testing.ino the sensor works again. Flashing Tasmota it doesn't work anymore.

@Jason2866
The ESP-breakout is on a breadboard. The TSL2561 works fine with Testing.ino. With Tasmota it is detected but never shows any data. I don't use MQTT with the lots of flashing to and fro. Just flashing the sketch with full erase -> generic device -> GPIO5/SCL GPIO4/SDA -> Reset.

The log of breakout R3.0 showed "Vcc":3.013i", so I suspected a power-supply problem and rearranged the wiring with R2.1 which supplied 3.5V. Still no joy. I also commented out all I2C-sensors in user_config.h except TSL2561. Same result.

joba-1, 2018-05-13:
Wild guess: maybe the esp sdk 2_4_1 is the problem? I see it in your version string(?) but not mine.
Maybe this uses too much ram or has an issue with wire or... ?

Jason2866, 2018-05-14:
@renne  My TSL was always detected too. I tried it again on the NodeMCU. I hooked the sca and scl pins
direct on the nodemcu and just wired the gnd and 3.3v pin. Now it works on the NodeMCU too.
So i guess it is on the edge with some specs. (Timing, voltage, ...)

renne, 2018-06-29:
I build it from the current github repo (git clone https://github.com/arendst/Sonoff-Tasmota.git).
Just flashed https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.14.0/sonoff-allsensors.bin onto the Electrodragon breakout and the H801 with
`esptool --port /dev/ttyUSB0 erase_flash && esptool --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 sonoff-allsensors.bin`.

After that I configured the module in generic mode and two GPIOs as SCL and SDA. On both devices I2CScan shows the TSL2561 on 0x29h but nothing about the TSL2561 in the teleperiod message or web-interface.

renne, 2018-06-29:
A few minutes ago I did a `git clone https://github.com/arendst/Sonoff-Tasmota.git`, uncommented `  #define USE_TSL2561`, set `#define WEB_LOG_LEVEL          LOG_LEVEL_DEBUG_MORE` and
```
#define STA_SSID1              "<MySSID>"                // [Ssid1] Wifi SSID
#define STA_PASS1              "<MyPSK>"                // [Password1] Wifi password
```
and flashed it on the H801 with the Arduino IDE. Afterwards I configured generic module and SCL/SDA GPIOs again.

This is the result:
```
00:00:00 CFG: Loaded from flash at F4, Count 8
00:00:00 APP: Boot Count 5
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_6F9563, GroupTopic sonoffs) Version 6.0.0b-2_4_1
00:00:00 CFG: Saved to flash at FB, Count 9, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 BartschNet in mode 11N as sonoff-5475...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Checking connection...
00:00:07 WIF: Connected
00:00:07 DNS: Initialized
00:00:07 HTP: Web server active on sonoff-5475.local with IP address 192.168.100.47
00:00:07 HTP: Main Menu
00:00:08 MQT: Attempting connection...
00:00:09 DNS: Query done. MQTT services found 0
00:00:09 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
00:00:10 APP: (UTC) Fri Jun 29 13:57:34 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
14:57:40 HTP: Configuration
14:57:41 HTP: Configure Module
14:57:44 MQT: Attempting connection...
14:57:45 DNS: Query done. MQTT services found 0
14:57:45 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
14:57:46 HTP: Configuration
14:57:49 HTP: Main Menu
14:57:50 HTP: Console
14:57:53 WIF: Checking connection...
14:57:53 WIF: Connected
14:57:56 MQT: Attempting connection...
14:57:57 DNS: Query done. MQTT services found 0
14:57:57 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
14:58:08 MQT: Attempting connection...
14:58:09 DNS: Query done. MQTT services found 0
14:58:09 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
14:58:09 CMD: i2cscan
14:58:09 SRC: WebConsole from 192.168.100.30
14:58:09 RSL: Received Topic /i2cscan, Data Size 0, Data 
14:58:09 RSL: Group 0, Index 1, Command I2CSCAN, Data 
14:58:09 RSL: stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x29"}
14:58:14 WIF: Checking connection...
14:58:14 WIF: Connected
14:58:20 MQT: Attempting connection...
14:58:21 DNS: Query done. MQTT services found 0
14:58:21 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
14:58:32 MQT: Attempting connection...
14:58:33 DNS: Query done. MQTT services found 0
14:58:33 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
14:58:36 WIF: Checking connection...
14:58:36 WIF: Connected
14:58:44 MQT: Attempting connection...
14:58:45 DNS: Query done. MQTT services found 0
14:58:45 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
14:58:56 MQT: Attempting connection...
14:58:57 DNS: Query done. MQTT services found 0
14:58:57 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
14:58:58 WIF: Checking connection...
14:58:58 WIF: Connected
14:59:08 MQT: Attempting connection...
14:59:09 DNS: Query done. MQTT services found 0
14:59:09 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
14:59:19 WIF: Checking connection...
14:59:19 WIF: Connected
14:59:19 MQT: Attempting connection...
14:59:20 DNS: Query done. MQTT services found 0
14:59:20 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
14:59:31 MQT: Attempting connection...
14:59:32 DNS: Query done. MQTT services found 0
14:59:32 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
14:59:41 WIF: Checking connection...
14:59:41 WIF: Connected
14:59:43 MQT: Attempting connection...
14:59:44 DNS: Query done. MQTT services found 0
14:59:44 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
```

Jason2866, 2018-06-29:
You have a general problem. You get no mqtt connection.

renne, 2018-06-29:
I usually test without the MQTT server configured to avoid side effects. A sensor should work without MQTT enabled. After setting the MQTT server IP address the TSL2561 is detected but there seems to be a communications problem:

```
00:00:00 CFG: Loaded from flash at F9, Count 11
00:00:00 APP: Boot Count 7
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_6F9563, GroupTopic sonoffs) Version 6.0.0b-2_4_1
00:00:00 CFG: Saved to flash at F8, Count 12, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 BartschNet in mode 11N as sonoff-5475...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Checking connection...
00:00:07 WIF: Connected
00:00:07 DNS: Initialized
00:00:07 HTP: Web server active on sonoff-5475.local with IP address 192.168.100.47
00:00:07 APP: (UTC) Fri Jun 29 14:40:20 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
15:40:21 MQT: Attempting connection...
15:40:21 MQT: Connected
15:40:21 MQT: tele/sonoff/LWT = Online (retained)
15:40:21 MQT: cmnd/sonoff/POWER = 
15:40:21 MQT: Subscribe to cmnd/sonoff/#
15:40:21 MQT: Subscribe to cmnd/sonoffs/#
15:40:21 MQT: Subscribe to cmnd/DVES_6F9563/#
15:40:21 MQT: tele/sonoff/INFO1 = {"Module":"Generic","Version":"6.0.0b","FallbackTopic":"DVES_6F9563","GroupTopic":"sonoffs"}
15:40:21 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5475","IPAddress":"192.168.100.47"}
15:40:21 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
15:40:23 HTP: Main Menu
15:40:27 HTP: Information
15:40:28 SHT: Sensor did not ACK command
15:40:28 I2C: TSL2561 found at 0x29
15:40:29 MQT: tele/sonoff/STATE = {"Time":"2018-06-29T15:40:29","Uptime":"0T00:00:16","Vcc":3.481,"Wifi":{"AP":1,"SSId":"BartschNet","RSSI":100,"APMac":"34:81:C4:2A:17:F3"}}
15:40:40 WIF: Checking connection...
15:40:40 WIF: Connected
15:40:47 HTP: Main Menu
15:40:49 HTP: Console
15:41:01 WIF: Checking connection...
15:41:01 WIF: Connected
15:41:21 WIF: Checking connection...
15:41:21 WIF: Connected
15:41:41 WIF: Checking connection...
15:41:41 WIF: Connected
15:42:01 WIF: Checking connection...
15:42:01 WIF: Connected
15:42:03 CMD: teleperiod 10
15:42:03 SRC: WebConsole from 192.168.100.30
15:42:03 RSL: Received Topic /teleperiod, Data Size 2, Data 10
15:42:03 RSL: Group 0, Index 1, Command TELEPERIOD, Data 10
15:42:03 MQT: stat/sonoff/RESULT = {"TelePeriod":"10"}
15:42:03 MQT: tele/sonoff/STATE = {"Time":"2018-06-29T15:42:03","Uptime":"0T00:01:50","Vcc":3.480,"Wifi":{"AP":1,"SSId":"BartschNet","RSSI":100,"APMac":"34:81:C4:2A:17:F3"}}
15:42:04 CFG: Saved to flash at F7, Count 13, Bytes 3584
15:42:05 HTP: Main Menu
15:42:10 HTP: Console
15:42:14 SHT: Sensor did not ACK command
15:42:15 MQT: tele/sonoff/STATE = {"Time":"2018-06-29T15:42:15","Uptime":"0T00:02:02","Vcc":3.483,"Wifi":{"AP":1,"SSId":"BartschNet","RSSI":100,"APMac":"34:81:C4:2A:17:F3"}}
15:42:23 WIF: Checking connection...
15:42:23 WIF: Connected
15:42:24 SHT: Sensor did not ACK command
15:42:25 MQT: tele/sonoff/STATE = {"Time":"2018-06-29T15:42:25","Uptime":"0T00:02:12","Vcc":3.484,"Wifi":{"AP":1,"SSId":"BartschNet","RSSI":100,"APMac":"34:81:C4:2A:17:F3"}}
15:42:35 SHT: Sensor did not ACK command
15:42:36 MQT: tele/sonoff/STATE = {"Time":"2018-06-29T15:42:36","Uptime":"0T00:02:23","Vcc":3.480,"Wifi":{"AP":1,"SSId":"BartschNet","RSSI":100,"APMac":"34:81:C4:2A:17:F3"}}
15:42:44 WIF: Checking connection...
15:42:44 WIF: Connected
15:42:46 SHT: Sensor did not ACK command
```
... and so on ...

joba-1, 2018-06-29:
Yes, but that could be normal for his setup and should not influence detecting sensors.

Status is: it does not work for renne with identical software that works for others. Means: it is hardware related. Still it is strange that i2cscan detects the device, but my lib (if it is called at all) does not when used within tasmota (but works with the examples on his hardware).

@renne : would you please retry with this function replaced in sonoff/xsns_16_tsl2561.ino:

`void Tsl2561Detect()
{
  if (!Tsl.available()) {
    snprintf_P(log_data, sizeof(log_data), S_LOG_I2C_FOUND_AT, "TSL !avail", Tsl.address());
    AddLog(LOG_LEVEL_DEBUG);
    Tsl.begin();
    if (Tsl.available()) {
      snprintf_P(log_data, sizeof(log_data), S_LOG_I2C_FOUND_AT, "TSL2561", Tsl.address());
      AddLog(LOG_LEVEL_DEBUG);
    }
  }
  else {
    snprintf_P(log_data, sizeof(log_data), S_LOG_I2C_FOUND_AT, "TSL avail", Tsl.address());
    AddLog(LOG_LEVEL_DEBUG);
  }
}
`
If my lib is called, then there must be some kind of output with debug level activated, no matter if the sensor is detected or not.

renne, 2018-06-29:
I added some debug messages to check out when the code in `xsns_16_tsl2561.ino` fails.
The method `Tsl2561Util::normalizedLuminosity(gain, exposure, full = scaledFull, ir = scaledIr)` returns `false` instead of `true` in line 63.

joba-1, 2018-06-29:
Ok, so my changes above are obsolete. Detection works now. Fast progress.

Since autogain must have returned true, you got values from the sensor. 
These values must be somehow out of range according to the data sheet, or plain zero all the time (or my code has a bug ... naaah). Can you please enter debug code that prints the parameters given to normalizedLuminosity? E.g. like this:

```
       && (Tsl2561Util::normalizedLuminosity(gain, exposure, full = scaledFull, ir = scaledIr), 1)
       && (Tsl2561Util::milliLux(full, ir, milliLux, Tsl2561::packageCS(id)), 1)) {

        snprintf_P(log_data, sizeof(log_data), PSTR(D_ILLUMINANCE " g:%d, e:%d, f:%u, i:%u -> %u.%03u " D_UNIT_LUX),
          gain, exposure, full, ir, scaledFull, scaledIr);
```

renne, 2018-06-29:
This is the console output with your code:
```
00:00:00 CFG: Loaded from flash at F5, Count 7
00:00:00 APP: Boot Count 4
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_6F9563, GroupTopic sonoffs) Version 6.0.0b-2_4_1
00:00:00 CFG: Saved to flash at F4, Count 8, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 BartschNet in mode 11N as sonoff-5475...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Checking connection...
00:00:07 WIF: Connected
00:00:07 DNS: Initialized
00:00:07 HTP: Web server active on sonoff-5475.local with IP address 192.168.100.47
00:00:07 APP: (UTC) Fri Jun 29 18:30:21 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
19:30:22 MQT: Attempting connection...
19:30:22 MQT: Connected
19:30:22 MQT: tele/sonoff/LWT = Online (retained)
19:30:22 MQT: cmnd/sonoff/POWER = 
19:30:22 MQT: Subscribe to cmnd/sonoff/#
19:30:22 MQT: Subscribe to cmnd/sonoffs/#
19:30:22 MQT: Subscribe to cmnd/DVES_6F9563/#
19:30:22 MQT: tele/sonoff/INFO1 = {"Module":"Generic","Version":"6.0.0b","FallbackTopic":"DVES_6F9563","GroupTopic":"sonoffs"}
19:30:22 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5475","IPAddress":"192.168.100.47"}
19:30:22 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
19:30:23 HTP: Main Menu
19:30:24 HTP: Console
19:30:29 SHT: Sensor did not ACK command
19:30:29 I2C: TSL2561 found at 0x29
19:30:30 MQT: tele/sonoff/STATE = {"Time":"2018-06-29T19:30:30","Uptime":"0T00:00:16","Vcc":3.483,"Wifi":{"AP":1,"SSId":"BartschNet","RSSI":98,"APMac":"34:81:C4:2A:17:F3"}}
19:30:31 Illuminance g:1, e:2, f:0, i:0 -> 0.000 lx
19:30:31 MQT: tele/sonoff/SENSOR = {"Time":"2018-06-29T19:30:30","TSL2561":{"Illuminance":0.000}}
19:30:40 SHT: Sensor did not ACK command
19:30:41 MQT: tele/sonoff/STATE = {"Time":"2018-06-29T19:30:41","Uptime":"0T00:00:27","Vcc":3.480,"Wifi":{"AP":1,"SSId":"BartschNet","RSSI":100,"APMac":"34:81:C4:2A:17:F3"}}
19:30:41 Illuminance g:1, e:2, f:0, i:0 -> 0.000 lx
19:30:41 MQT: tele/sonoff/SENSOR = {"Time":"2018-06-29T19:30:41","TSL2561":{"Illuminance":0.000}}
19:30:42 WIF: Checking connection...
19:30:42 WIF: Connected
19:30:50 SHT: Sensor did not ACK command
19:30:51 MQT: tele/sonoff/STATE = {"Time":"2018-06-29T19:30:51","Uptime":"0T00:00:37","Vcc":3.480,"Wifi":{"AP":1,"SSId":"BartschNet","RSSI":100,"APMac":"34:81:C4:2A:17:F3"}}
19:30:52 Illuminance g:1, e:2, f:0, i:0 -> 0.000 lx
```

joba-1, 2018-06-29:
This says raw sensor values at highest sensitivity are always zero. 
Looks like a defect sensor.
But it worked with my examples? Especially Autogain?

renne, 2018-06-30:
Output of Testing.ino:

Got luminosity after 419 ms. Full spectrum is 285 and IR only is
                                                                Testing Tsl2561 at address 29: found
                                                                                                    Chip has type 50 and revision 0
                                                                                                                                   Chip is a T, FN or CL type package
Chip powered on at 540707
                         New gain = 0, exposure = 0x02
                                                      Got luminosity after 420 ms. Full spectrum is 284 and IR only is 130
                                                                                                                          Chip powered on at 541128
                                                                                                                                                   New gain = 1, exposure = 0x02
                                                                                                                                                                                Got luminosity after 419 ms. F
ull spectrum is 4357 and IR only is 1990
                                        Chip powered on at 541548
                                                                 New gain = 0, exposure = 0x01
                                                                                              Got luminosity after 109 ms. Full spectrum is 78 and IR only is 36
                                                                                                                                                                Chip powered on at 541658
                                                                                                                                                                                         New gain = 1, exposur
e = 0x01
        Got luminosity after 109 ms. Full spectrum is 1087 and IR only is 496
                                                                             Chip powered on at 541768
                                                                                                      New gain = 0, exposure = 0x00
                                                                                                                                   Got luminosity after 24 ms. Full spectrum is 18 and IR only is 9
                                                                                                                                                                                                   Chip powere
d on at 541793
              New gain = 1, exposure = 0x00
                                           Got luminosity after 24 ms. Full spectrum is 137 and IR only is 63

Testing Tsl2561 at address 39: NOT found
                                                                                                                                                     Error was 2: Gone

Testing Tsl2561 at address 49: NOT found
                                        Error was 2: Gone

Next test in 5s

renne, 2018-06-30:
Output of Simple.ino:

Starting Tsl2561 simple loop
Tsl2561 at 0x29(id=0x50) luminosity is   140 (full) and    64 (ir)
                                                                  Tsl2561 at 0x29(id=0x50) luminosity is   141 (full) and    65 (ir)
                                                                                                                                    Tsl2561 at 0x29(id=0x50) luminosity is   141 (full) and    65 (ir)
                                                                                                                                                                                                      Tsl2561 
at 0x29(id=0x50) luminosity is   141 (full) and    65 (ir)
                                                          Tsl2561 at 0x29(id=0x50) luminosity is   141 (full) and    65 (ir)
                                                                                                                            Tsl2561 at 0x29(id=0x50) luminosity is   141 (full) and    65 (ir)

renne, 2018-06-30:
Output of Utility.ino:

                   Starting Tsl2561Util loop

Tsl2561 addr: 0x29, id: 0x50, sfull:    12, sir:     6, gain: 0, exp: 0, full:  5766, ir:  2956, lux:    37.930
                                                                                                               Tsl2561 addr: 0x29, id: 0x50, sfull:    66, sir:    30, gain: 0, exp: 1, full:  4356, ir:  2067
, lux:    37.915
                Tsl2561 addr: 0x29, id: 0x50, sfull:     0, sir:     0, gain: 0, exp: 2, full:     0, ir:     0: Tsl2561Util::normalizedLuminosity() error
                                                                                                                                                          Tsl2561 addr: 0x29, id: 0x50, sfull:   137, sir:    
63, gain: 1, exp: 0, full:  4156, ir:  1990, lux:    35.062
                                                           Tsl2561 addr: 0x29, id: 0x50, sfull:  1047, sir:   483, gain: 1, exp: 1, full:  4321, ir:  2079, lux:    35.833
                                                                                                                                                                          Tsl2561 addr: 0x29, id: 0x50, sfull:
     0, sir:     0, gain: 1, exp: 2, full:     0, ir:     0: Tsl2561Util::normalizedLuminosity() error

Tsl2561 addr: 0x29, id: 0x50, sfull:    12, sir:     6, gain: 0, exp: 0, full:  5766, ir:  2956, lux:    37.930
                                                                                                               Tsl2561 addr: 0x29, id: 0x50, sfull:    66, sir:    30, gain: 0, exp: 1, full:  4356, ir:  2067
, lux:    37.915
                Tsl2561 addr: 0x29, id: 0x50, sfull:     0, sir:     0, gain: 0, exp: 2, full:     0, ir:     0: Tsl2561Util::normalizedLuminosity() error
                                                                                                                                                          Tsl2561 addr: 0x29, id: 0x50, sfull:   137, sir:    
63, gain: 1, exp: 0, full:  4156, ir:  1990, lux:    35.062
                                                           Tsl2561 addr: 0x29, id: 0x50, sfull:  1046, sir:   483, gain: 1, exp: 1, full:  4317, ir:  2079, lux:    35.680
                                                                                                                                                                          Tsl2561 addr: 0x29, id: 0x50, sfull:
     0, sir:     0, gain: 1, exp: 2, full:     0, ir:     0: Tsl2561Util::normalizedLuminosity() error

renne, 2018-06-30:
Output of Autogain.ino:

Starting Tsl2561Util autogain loop
Tsl2561 addr: 0x29, id: 0x50, sfull:  3350, sir:  1660, full:  3350, ir:  1660, gain: 1, exp: 2, lux:    24.787
                                                                                                               Tsl2561 addr: 0x29, id: 0x50, sfull:  3377, sir:  1675, full:  3377, ir:  1675, gain: 1, exp: 2, lux:    24.886
                Tsl2561 addr: 0x29, id: 0x50, sfull:  3379, sir:  1677, full:  3379, ir:  1677, gain: 1, exp: 2, lux:    24.839
                                                                                                                               Tsl2561 addr: 0x29, id: 0x50, sfull:  3377, sir:  1676, full:  3377, ir:  1676, gain: 1, exp: 2, lux:    24.824
                                Tsl2561 addr: 0x29, id: 0x50, sfull:  3378, sir:  1676, full:  3378, ir:  1676, gain: 1, exp: 2, lux:    24.863
                                                                                                                                               Tsl2561 addr: 0x29, id: 0x50, sfull:  3377, sir:  1675, full:  3377, ir:  1675, gain: 1, exp: 2, lux:    24.886
                                                Tsl2561 addr: 0x29, id: 0x50, sfull:  3376, sir:  1675, full:  3376, ir:  1675, gain: 1, exp: 2, lux:    24.848
                                                                                                                                                               Tsl2561 addr: 0x29, id: 0x50, sfull:  3376, sir:  1674, full:  3376, ir:  1674, gain: 1, exp: 2, lux:    24.910
                                                                Tsl2561 addr: 0x29, id: 0x50, sfull:  3375, sir:  1674, full:  3375, ir:  1674, gain: 1, exp: 2, lux:    24.872
                                                                                                                                                                               Tsl2561 addr: 0x29, id: 0x50, sfull:  3377, sir:  1675, full:  3377, ir:  1675, gain: 1, exp: 2, lux:    24.886

renne, 2018-06-30:
`default:               delay(420); in waitNext()` does not work but 430 did the trick.
```
17:04:55 Illuminance g:1, e:2, f:4678, i:1549 -> 78.999 lx
17:04:55 MQT: tele/sonoff/SENSOR = {"Time":"2018-06-30T17:04:55","TSL2561":{"Illuminance":78.999}}
```

joba-1, 2018-07-08:
This release increases the wait time to 430ms and should solve this issue: 
https://github.com/joba-1/Joba_Tsl2561/releases/tag/v2.0.7
Should be picked up by PlatformIO and Arduino IDE library managers soon.

ascillato2, 2018-07-10:
Hi,

Could you managed to solve this issue with last version?

ascillato2, 2018-07-12:
Great!

Closing this issue then. Please, reopen it if needed. Thanks!

closed 2018-07-12 by ascillato2