arendst/Tasmota #2417: Unable to see tele/  messages in MQTT

LEMD49, 2018-04-11:
Having updated to 5.12.0j and changed the hardware from sonoff basic to sonoff POW, I now found that my **MQTT server** and therefore NODE-RED flows are ### **no longer able to see the tele/sonoff/STATE messages** and therefore unable to have a proper dashboard functionality. The cmnd and stat messages work fine. 
In the console for some reason, the tele items appear with RSL instead of MQT field.  How to remedy?

Make sure these boxes are checked before submitting your issue - Thank you!

- [x] What hardware you are using
sonoff POW
macmini server running MOSQUITTO as a service and NODE-RED

- [x] Provide the output of command ``status 0``
08:02:19 CMD: status 0
08:02:19 RSL: stat/gatelights/STATUS = {"Status":{"Module":6,"FriendlyName":["gatelights"],"Topic":"gatelights","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
08:02:19 RSL: stat/gatelights/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://10.0.8.50/arduino/sonoff/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T10:10:23","StartupUTC":"2018-04-10T20:51:56","Sleep":0,"BootCount":12,"SaveCount":60,"SaveAddress":"F5000"}}
08:02:19 RSL: stat/gatelights/STATUS2 = {"StatusFWR":{"Version":"5.12.0j","BuildDateTime":"2018-04-10T14:58:31","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
08:02:19 RSL: stat/gatelights/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"etienne","LogPort":514,"SSId":["server","ATC2AE"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
08:02:19 RSL: stat/gatelights/STATUS4 = {"StatusMEM":{"ProgramSize":502,"Free":436,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
08:02:19 RSL: stat/gatelights/STATUS5 = {"StatusNET":{"Hostname":"gatelights-4352","IPAddress":"10.0.8.51","Gateway":"10.0.8.50","Subnetmask":"255.255.255.0","DNSServer":"10.0.8.1","Mac":"5C:CF:7F:92:D1:00","Webserver":2,"WifiConfig":3}}
08:02:19 RSL: stat/gatelights/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.8.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_92D100","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:02:19 RSL: stat/gatelights/STATUS7 = {"StatusTIM":{"UTC":"Wed Apr 11 07:02:19 2018","Local":"Wed Apr 11 08:02:19 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:41","Sunset":"19:48"}}
08:02:19 RSL: stat/gatelights/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
08:02:19 RSL: stat/gatelights/STATUS10 = {"StatusSNS":{"Time":"2018-04-11T08:02:19","ENERGY":{"Total":0.038,"Yesterday":0.037,"Today":0.001,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
08:02:19 RSL: stat/gatelights/STATUS11 = {"StatusSTS":{"Time":"2018-04-11T08:02:19","Uptime":"0T10:10:23","Vcc":3.552,"POWER":"OFF","Wifi":{"AP":2,"SSId":"ATC2AE","RSSI":44,"APMac":"20:C9:D0:AB:6F:AC"}}}

EOM

curzon01, 2018-04-11:
- Check MQTT Server (10.0.8.50 ?) username (DVES_USER ?) and password for mosquitto on Web GUI "Configuration" -> "Configure MQTT"
- Goto Web GUI "Configuration" -> "Configure Logging" and set "Web Log Level" to "2 Info"
- Observe the console MQTT connection, it should be shown as
MQT: Attempting connection...
MQT: Connected

LEMD49, 2018-04-11:
1. check OK
2. was 2 Info check ok
3. getting MQT: connected, receiving find all  cmnd and stat messages but not tele as they are RSL type and not MQT, so no longer captured by MQTT broker. 

All I changed was the HW from basic to POW and version

Extract of console below.
The MQT: stat/+/+ are seen in my broker and Nodered by the RSL: tele/+/+ are not.
Previously the tele/+/+ were also of MQT type but after HW and version change no longer?
TKs

11:37:39 MQT: stat/gatelights/RESULT = {"POWER":"OFF"}
11:37:39 MQT: stat/gatelights/POWER = OFF
11:38:39 MQT: stat/gatelights/RESULT = {"POWER":"OFF"}
11:38:39 MQT: stat/gatelights/POWER = OFF
11:39:39 MQT: stat/gatelights/RESULT = {"POWER":"OFF"}
11:39:39 MQT: stat/gatelights/POWER = OFF
11:40:00 RSL: tele/gatelights/STATE = {"Time":"2018-04-11T11:40:00","Uptime":"0T00:25:13","Vcc":3.552,"POWER":"OFF","Wifi":{"AP":2,"SSId":"ATC2AE","RSSI":40,"APMac":"20:C9:D0:AB:6F:AC"}}
11:40:00 RSL: tele/gatelights/SENSOR = {"Time":"2018-04-11T11:40:00","ENERGY":{"Total":0.038,"Yesterday":0.037,"Today":0.001,"Period":0,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}

LEMD49, 2018-04-11:
in other words why do the tele/... messages appear as LOG_RESULT (RSL) in lieu of LOG_MQTT (MQT) messages and how to change that (back to MQT)?

RaymondMouthaan, 2018-04-11:
Have you checked the status of mqtt with SetOption3 and SetOption3 1 to enable it?

curzon01, 2018-04-12:
two other steps to try:

1. try same device you have with exact this config with downgraded FW 5.12.0 from [release repository](https://github.com/arendst/Sonoff-Tasmota/releases)
2. if this works, try compile your developer version 5.12.0j using lib 2.3.0 instead of 2.4.1

LEMD49, 2018-04-12:
yes!  after rolling back to FW5.12.0 issue is gone as my tele/+/+ messages are back as MQTT and the console confirms this as they are back to MQT and no longer RSL.
Many tkx!

20:07:15 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
20:07:15 MQT: stat/sonoff/POWER = OFF
20:07:15 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
20:07:15 MQT: stat/sonoff/POWER = OFF
20:07:22 MQT: tele/sonoff/STATE = {"Time":"2018-04-12T20:07:22","Uptime":"0T00:00:14","Vcc":3.546,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ATC2","RSSI":100,"APMac":"00:21:E9:B7:4F:7C"}}
20:07:22 MQT: tele/sonoff/SENSOR = {"Time":"2018-04-12T20:07:22","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}

curzon01, 2018-04-13:
if you can live with the release version pls close issue - otherwise try to compile your own build with lib 2.3.0 instead 2.4.1

closed 2018-04-13 by LEMD49