arendst/Tasmota #2419: tasmota reconnects every 30 minutes

mizrachiran, 2018-04-11:
Hi,

I'm using the latest Tasmota version on Esp-12 module.
I noticed that all of my devices reconnects to my mqtt broker every 30 minutes (exactly).
no sure if it disconnect from the wifi or only from the mqtt broker.

Is this a bug or a feature in the firmware? Can I control this behavior?

- [x ] What hardware you are using Generic esp-12
- [] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

Jason2866, 2018-04-11:
It is not a feature and not a general bug.
I have version 5.12.0k on my test device installed yesterday and no reconnect since that time

RaymondMouthaan, 2018-04-11:
Never seen this behaviour with my 4 sonoff and 6 wemos devices. Possibly your broker drops the connection every 30 minutes or some reason (wifi) network has issues ... Try a mqtt client other than Tasmota from your pc and check whether it looses it's connection too. 
Another thing you could try is to use a different MQTT library type. The one below is the one I am using and works perfectly.

user_config.h
```
#define MQTT_LIBRARY_TYPE      MQTT_TASMOTAMQTT 
```

mizrachiran, 2018-04-11:
I thought it’s the broker. However, it doesnt happen with other devices.

mizrachiran, 2018-04-15:
Adding some information, I see the following lines every 30 minutes in the console:

21:19:07 MQT: Attempting connection...
21:19:07 MQT: Connect failed to 10.0.0.30:1883, rc -2. Retry in 10 sec
21:19:19 MQT: Attempting connection...
21:19:19 MQT: Connected

**My full device status output**

{
"Status": {
"Module": 18,
"FriendlyName": "Sonoff",
"Topic": "luxo-2",
"ButtonTopic": "0",
"Power": 0,
"PowerOnState": 3,
"LedState": 1,
"SaveData": 1,
"SaveState": 1,
"ButtonRetain": 0,
"PowerRetain": 0
},
"StatusPRM": {
"Baudrate": 115200,
"GroupTopic": "sonoffs",
"OtaUrl": "http://domus1:80/api/arduino/sonoff.ino.bin",
"Uptime": "1T05:01:52",
"Sleep": 0,
"BootCount": 13,
"SaveCount": 100,
"SaveAddress": "F8000"
},
"StatusFWR": {
"Version": "5.12.0",
"BuildDateTime": "2018-02-09T14:54:51",
"Boot": 31,
"Core": "2_3_0",
"SDK": "1.5.3(aec24ac9)"
},
"StatusLOG": {
"SerialLog": 0,
"WebLog": 2,
"SysLog": 0,
"LogHost": "domus1",
"LogPort": 514,
"SSId1": "Mizrachi_2.4Ghz",
"SSId2": "Mizrachi_2.4Ghz",
"TelePeriod": 180,
"SetOption": "00000009"
},
"StatusMEM": {
"ProgramSize": 485,
"Free": 516,
"Heap": 21,
"ProgramFlashSize": 1024,
"FlashSize": 4096,
"FlashMode": 255
},
"StatusNET": {
"Hostname": "luxo-2",
"IPAddress": "10.0.0.102",
"Gateway": "10.0.0.1",
"Subnetmask": "255.255.255.0",
"DNSServer": "82.102.139.10",
"Mac": "5C:CF:7F:69:4A:C2",
"Webserver": 2,
"WifiConfig": 2
},
"StatusMQT": {
"MqttHost": "10.0.0.30",
"MqttPort": 1883,
"MqttClientMask": "luxo-2",
"MqttClient": "luxo-2",
"MqttUser": "DVES_USER",
"MAX_PACKET_SIZE": 1000,
"KEEPALIVE": 15
},
"StatusTIM": {
"UTC": "Sun Apr 15 18:41:56 2018",
"Local": "Sun Apr 15 21:41:56 2018",
"StartDST": "Sun Mar 25 02:00:00 2018",
"EndDST": "Sun Oct 28 03:00:00 2018",
"Timezone": 3
},
"StatusSNS": {
"Time": "2018-04-15T21:41:56",
"Switch1": "ON",
"Switch2": "OFF",
"Switch3": "ON"
},
"StatusSTS": {
"Time": "2018-04-15T21:41:56",
"Uptime": "1T05:01:52",
"Vcc": 1.248,
"POWER1": "OFF",
"POWER2": "OFF",
"POWER3": "OFF",
"Wifi": {
"AP": 1,
"SSId": "Mizrachi_2.4Ghz",
"RSSI": 76,
"APMac": "74:DA:DA:42:D0:52"
}
}
}

Jason2866, 2018-04-15:
Some suggestions
Just use one SSID if the are identic, no internal DNS Server? Most routers do this job.
Try to use it with one relay definition. You have defined power1 power2 power3

mizrachiran, 2018-04-15:
Thanks Jason, I will give it a try.