arendst/Tasmota #2420: Empty FriendlyName in STATUS command

RaymondMouthaan, 2018-04-11:
Make sure these boxes are checked before submitting your issue - Thank you!

- [x] What hardware you are using: Wemos D1 Mini v3.0.0
- [x] What firmware you are using: Sonoff-Tasmota 5.12.0k
- [x] Provide the output of command ``status 0``

```
16:26:16 MQT: stat/wemos-6/STATUS = {"Status":{"Module":18,"FriendlyName":[],"Topic":"wemos-6","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
16:26:16 MQT: stat/wemos-6/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"wemoss","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:27","StartupUTC":"2018-04-11T15:22:49","Sleep":0,"BootCount":11,"SaveCount":23,"SaveAddress":"3F5000"}}
16:26:16 MQT: stat/wemos-6/STATUS2 = {"StatusFWR":{"Version":"5.12.0k","BuildDateTime":"2018-04-11T16:57:28","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
16:26:16 MQT: stat/wemos-6/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["indonesia23",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
16:26:16 MQT: stat/wemos-6/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":2544,"Heap":17,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3}}
16:26:16 MQT: stat/wemos-6/STATUS5 = {"StatusNET":{"Hostname":"wemos-6-1923","IPAddress":"192.168.2.231","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.10","Mac":"DC:4F:22:1D:87:83","Webserver":2,"WifiConfig":3}}
16:26:16 MQT: stat/wemos-6/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.indonesia","MqttPort":1883,"MqttClientMask":"DVES_W6_%06X","MqttClient":"DVES_W6_1D8783","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:26:16 MQT: stat/wemos-6/STATUS7 = {"StatusTIM":{"UTC":"Wed Apr 11 15:26:16 2018","Local":"Wed Apr 11 16:26:16 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:07","Sunset":"19:35"}}
16:26:16 MQT: stat/wemos-6/STATUS10 = {"StatusSNS":{"Time":"2018-04-11T16:26:16","SHT3X-0x45":{"Temperature":25.5,"Humidity":43.7},"TempUnit":"C"}}
16:26:16 MQT: stat/wemos-6/STATUS11 = {"StatusSTS":{"Time":"2018-04-11T16:26:16","Uptime":"0T00:03:27","Vcc":2.993,"Wifi":{"AP":1,"SSId":"xxxxxxxxxxx","RSSI":100,"APMac":"30:D3:2D:70:05:4D"}}}
```

In the STATUS json the array FriendlyName is empty but when executing command FriendlyName for console it shows:

```
16:31:55 CMD: FriendlyName
16:31:55 MQT: stat/wemos-6/RESULT = {"FriendlyName1":"Wemos Six"}
```

This friendlyname has been set via web and via console and shows correctly after executing above command and on the web interface (see screenshot), but shows empty in the STATUS json, which is used by sonweb to adopt.

[image]

arendst, 2018-04-11:
Pls provide output of command ``cfgdump 3a0 10``

RaymondMouthaan, 2018-04-11:
Here you go ...

```
16:45:44 CMD: cfgdump 3a0 10
16:45:44 3A0:  FF 00 00 00  08 00 01 00  01 00 00 00  57 65 6D 6F |            Wemo|
16:45:44 3B0:  73 20 53 69  78 00 22 00  00 00 00 00  00 00 00 00 |s Six "         |
16:45:44 3C0:  00 00 00 00  00 00 00 00  00 00 00 00  00 53 6F 6E |             Son|
16:45:44 3D0:  6F 66 66 32  00 00 00 00  00 00 00 00  00 00 00 00 |off2            |
16:45:44 3E0:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 53 6F |              So|
16:45:44 3F0:  6E 6F 66 66  33 00 00 00  00 00 00 00  00 00 00 00 |noff3           |
16:45:44 400:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 53 |               S|
16:45:44 410:  6F 6E 6F 66  66 34 00 00  00 00 00 00  00 00 00 00 |onoff4          |
16:45:44 420:  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |                |
16:45:44 430:  30 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00 |0               |
16:45:44 MQT: stat/wemos-6/RESULT = {"CfgDump":"Done"}
```

arendst, 2018-04-11:
Looks fine to me. Strange as mine shows friendlyname as expected:
```
17:48:19 MQT: stat/ring2/STATUS = {"Status":{"Module":18,"FriendlyName":["Wemos Six"],"Topic":"ring2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
```
Ah, if you do not configure a device you won't see the friendlyname. I'll have a look as I thought I changed it for both webserver main and information page but apparently forgot the status.

RaymondMouthaan, 2018-04-11:
Theo,

This one (Wemos Five) looks fine, but it runs version H 
```
17:53:24 CMD: STATUS 0
17:53:24 MQT: stat/wemos-5/STATUS = {"Status":{"Module":18,"FriendlyName":"Wemos Five","Topic":"wemos-5","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
17:53:24 MQT: stat/wemos-5/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"wemoss","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Hardware Watchdog","Uptime":"1T02:50:57","StartupUTC":"2018-04-10T13:02:27","Sleep":0,"BootCount":44,"SaveCount":23192,"SaveAddress":"3F4000"}}
17:53:24 MQT: stat/wemos-5/STATUS2 = {"StatusFWR":{"Version":"5.12.0h","BuildDateTime":"2018-03-24T19:16:14","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
17:53:24 MQT: stat/wemos-5/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId1":"indonesia23","SSId2":"","TelePeriod":30,"SetOption":"00000009"}}
17:53:24 MQT: stat/wemos-5/STATUS4 = {"StatusMEM":{"ProgramSize":506,"Free":2564,"Heap":18,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3}}
17:53:24 MQT: stat/wemos-5/STATUS5 = {"StatusNET":{"Hostname":"wemos-5-4988","IPAddress":"192.168.2.229","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.10","Mac":"DC:4F:22:1C:93:7C","Webserver":2,"WifiConfig":3}}
17:53:24 MQT: stat/wemos-5/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.indonesia","MqttPort":1883,"MqttClientMask":"DVES_W5_%06X","MqttClient":"DVES_W5_1C937C","MqttUser":"DVES_USER","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:53:24 MQT: stat/wemos-5/STATUS7 = {"StatusTIM":{"UTC":"Wed Apr 11 15:53:24 2018","Local":"Wed Apr 11 17:53:24 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
17:53:24 MQT: stat/wemos-5/STATUS10 = {"StatusSNS":{"Time":"2018-04-11T17:53:24","TSL2561":{"Illuminance":4087.874}}}
17:53:24 MQT: stat/wemos-5/STATUS11 = {"StatusSTS":{"Time":"2018-04-11T17:53:24","Uptime":"1T02:50:57","Vcc":3.039,"Wifi":{"AP":1,"SSId":"zzzzzzzzzzz","RSSI":100,"APMac":"30:D3:2D:70:05:4D"}}}
```

Ill try the above suggestion after diner and let you know the results 👍

RaymondMouthaan, 2018-04-11:
@arendst after your modifications to sonoff.ino it works perfect 🥇, thanks!

closed 2018-04-11 by RaymondMouthaan