arendst/Tasmota #2423: Platformio OTA Upload to webserver failing

ScottyDS, 2018-04-12:
Not a Tasmota Issue! 

Please could someone advise me on where to look to fix the below error? I have scoured the web and cant find an answer. pycurl is installed, and the upload works once after a re-install of Atom, Python and Platformio, and configuration of platformio.ini.  (I am using Windows 10)

I then configure the user settings.h, and when I try upload to the web server again I get the following error.
Uploading .pioenvs\sonoff\firmware.bin
Traceback (most recent call last):
File "F:\Platformio\Sonoff-Tasmota-development\pio\espupload.py", line 16, in <module>
import pycurl
ImportError: No module named pycurl

Geitde, 2018-04-12:
i never did that, but the problem seems to be your upload tool, not Tasmota itself. The python module pycurl is missing or not found on your computer.

ScottyDS, 2018-04-12:
That is what I initially thought. And that is why I reinstalled everything. But the issue only occurs when I edit the user_config.h. When I restore the defaults to the userconfig.h, the upload works again. Have been struggling with this for 3 days now.
 I will attempt the same on my Linux machine this evening and if the issue does not occur there, I will close the issue.

closed 2018-04-26 by ScottyDS