arendst/Tasmota #2424: PowerOnState 3 seems not working

dh-harald, 2018-04-12:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
Sonoff Touch (EU version)
- [ ] Provide the output of command ``status 0``
```08:33:50 MQT: home/sonoff/stat/wsw01/STATUS = {"Status":{"Module":10,"FriendlyName":"wsw01","Topic":"wsw01","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
08:33:50 MQT: home/sonoff/stat/wsw01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://xxxx/sonoff/firmware.bin","Uptime":"0T17:29:04","Sleep":100,"BootCount":360,"SaveCount":8782,"SaveAddress":"FA000"}}
08:33:50 MQT: home/sonoff/stat/wsw01/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-10T05:57:16","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
08:33:50 MQT: home/sonoff/stat/wsw01/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"SysLog":3,"LogHost":"xxxx","LogPort":514,"SSId1":"xxxx","SSId2":"xxxx","TelePeriod":300,"SetOption":"55800029"}}
08:33:50 MQT: home/sonoff/stat/wsw01/STATUS4 = {"StatusMEM":{"ProgramSize":493,"Free":508,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
08:33:50 MQT: home/sonoff/stat/wsw01/STATUS5 = {"StatusNET":{"Hostname":"wsw01-0262","IPAddress":"xxxx","Gateway":"xxxx","Subnetmask":"255.255.255.0","DNSServer":"xxxx","Mac":"xxxx","Webserver":2,"WifiConfig":3}}
08:33:50 MQT: home/sonoff/stat/wsw01/STATUS6 = {"StatusMQT":{"MqttHost":"xxxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_854106","MqttUser":"DVES_wsw01","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:33:50 MQT: home/sonoff/stat/wsw01/STATUS7 = {"StatusTIM":{"UTC":"Thu Apr 12 08:33:50 2018","Local":"Thu Apr 12 08:33:50 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":0}}
08:33:50 MQT: home/sonoff/stat/wsw01/STATUS10 = {"StatusSNS":{"Time":"2018-04-12T08:33:50"}}
08:33:50 MQT: home/sonoff/stat/wsw01/STATUS11 = {"StatusSTS":{"Time":"2018-04-12T08:33:50","Uptime":"0T17:29:04","Vcc":3.580,"POWER":"ON","Wifi":{"AP":1,"SSId":"VM9297705","RSSI":100,"APMac":"6C:3B:6B:75:29:BA"}}}```
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

I'm using Sonoff Touch switches with almost default configuration. I facing that issue, if my switch disconnects from wifi, it doesn't pick up the last powered state. It came up with powered off state...
Here's a sample log:
```Apr 11 21:10:00 wsw01-0262 ESP-APP: Button1 multi-press 1
Apr 11 21:10:01 wsw01-0262 ESP-MQT: home/sonoff/stat/wsw01/RESULT = {"POWER":"ON"}
Apr 11 21:10:01 wsw01-0262 ESP-MQT: home/sonoff/stat/wsw01/POWER = ON (retained)
Apr 11 21:10:02 wsw01-0262 ESP-CFG: Saved to flash at F6, Count 8762, Bytes 1648
Apr 11 21:12:30 wsw01-0262 ESP-MQT: home/sonoff/tele/wsw01/STATE = {"Time":"2018-04-11T20:12:30","Uptime":"0T11:20:14","Vcc":3.591,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxx","RSSI":98,"A
PMac":"xxx"}}
Apr 11 21:22:33 wsw01-0262 ESP-RSL: home/sonoff/tele/wsw01/STATE = {"Time":"2018-04-11T20:22:33","Uptime":"0T11:25:14","Vcc":3.578,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxx","RSSI":100,"
APMac":"xxx"}}
Apr 11 21:22:35 wsw01-0262 ESP-DNS: Initialized
Apr 11 21:22:35 wsw01-0262 ESP-MQT: Attempting connection...
Apr 11 21:22:35 wsw01-0262 ESP-MQT: Connected
Apr 11 21:22:35 wsw01-0262 ESP-MQT: home/sonoff/tele/wsw01/LWT = Online (retained)
Apr 11 21:22:35 wsw01-0262 ESP-MQT: home/sonoff/cmnd/wsw01/POWER = 
Apr 11 21:22:35 wsw01-0262 ESP-MQT: Subscribe to home/sonoff/cmnd/wsw01/#
Apr 11 21:22:35 wsw01-0262 ESP-MQT: Subscribe to home/sonoff/cmnd/sonoffs/#
Apr 11 21:22:35 wsw01-0262 ESP-MQT: Subscribe to cmnd/DVES_854106/#
Apr 11 21:22:35 wsw01-0262 ESP-RSL: Group 0, Index 1, Command POWER, Data OFF
Apr 11 21:22:35 wsw01-0262 ESP-MQT: home/sonoff/stat/wsw01/RESULT = {"POWER":"OFF"}
Apr 11 21:22:35 wsw01-0262 ESP-MQT: home/sonoff/stat/wsw01/POWER = OFF (retained)
Apr 11 21:22:36 wsw01-0262 ESP-CFG: Saved to flash at F5, Count 8763, Bytes 1648
```
Any ideas, what's wrong?

reloxx13, 2018-04-12:
> disconnects from wifi
power on is not wifi connected.

poweron is triggered after power loss, thats why its called poweron and not wificonnected...

dh-harald, 2018-04-12:
Oh, I see... is it any parameter, to get this functionality on wifi reconnect?

reloxx13, 2018-04-12:
nope, cuz normally their is no reconnect.

also the state does not get lost if the wifi disconnects.

relais on => wifi off => relais (stays) on

relais on => power loss => relais off => power on => relais on again

dh-harald, 2018-04-12:
Its quiet interesting, because I've got 2 S20 switches, and those are working correctly...
I've restarted the mqtt server, results:
Touch:
```
Apr 12 11:52:13 wsw02-3662 ESP-RSL: home/sonoff/tele/wsw02/STATE = {"Time":"2018-04-12T10:52:14","Uptime":"0T18:35:17","Vcc":3.473,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxx","RSSI":100,"APMac":"xxx"}}
Apr 12 11:52:27 wsw02-3662 ESP-MQT: Attempting connection...
Apr 12 11:52:27 wsw02-3662 ESP-MQT: Connected
Apr 12 11:52:27 wsw02-3662 ESP-MQT: home/sonoff/tele/wsw02/LWT = Online (retained)
Apr 12 11:52:27 wsw02-3662 ESP-MQT: home/sonoff/cmnd/wsw02/POWER = 
Apr 12 11:52:27 wsw02-3662 ESP-MQT: Subscribe to home/sonoff/cmnd/wsw02/#
Apr 12 11:52:27 wsw02-3662 ESP-MQT: Subscribe to home/sonoff/cmnd/sonoffs/#
Apr 12 11:52:27 wsw02-3662 ESP-MQT: Subscribe to cmnd/DVES_84CE4E/#
Apr 12 11:52:27 wsw02-3662 ESP-RSL: Group 0, Index 1, Command POWER, Data OFF
Apr 12 11:52:27 wsw02-3662 ESP-MQT: home/sonoff/stat/wsw02/RESULT = {"POWER":"OFF"}
Apr 12 11:52:27 wsw02-3662 ESP-MQT: home/sonoff/stat/wsw02/POWER = OFF (retained)
Apr 12 11:52:28 wsw02-3662 ESP-CFG: Saved to flash at FA, Count 9511, Bytes 1648
```
meanwhile S20:
```
Apr 12 11:43:23 acsw01-3515 ESP-MQT: home/sonoff/tele/acsw01/STATE = {"Time":"2018-04-12T11:43:19","Uptime":"1T07:20:12","Vcc":3.528,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxx","RSSI":100,"APMac":"xxx"}}
Apr 12 11:52:05 acsw01-3515 ESP-MQT: Attempting connection...
Apr 12 11:52:05 acsw01-3515 ESP-MQT: Connect failed to xxx:1883, rc -2. Retry in 10 sec
Apr 12 11:53:00 acsw01-3515 ESP-MQT: Attempting connection...
Apr 12 11:53:00 acsw01-3515 ESP-MQT: Connected
Apr 12 11:53:00 acsw01-3515 ESP-MQT: home/sonoff/tele/acsw01/LWT = Online (retained)
Apr 12 11:53:00 acsw01-3515 ESP-MQT: home/sonoff/cmnd/acsw01/POWER = 
Apr 12 11:53:00 acsw01-3515 ESP-MQT: Subscribe to home/sonoff/cmnd/acsw01/#
Apr 12 11:53:00 acsw01-3515 ESP-MQT: Subscribe to home/sonoff/cmnd/sonoffs/#
Apr 12 11:53:00 acsw01-3515 ESP-MQT: Subscribe to cmnd/DVES_acsw01/#
Apr 12 11:53:00 acsw01-3515 ESP-RSL: Group 0, Index 1, Command POWER, Data ON
Apr 12 11:53:00 acsw01-3515 ESP-MQT: home/sonoff/stat/acsw01/RESULT = {"POWER":"ON"}
Apr 12 11:53:00 acsw01-3515 ESP-MQT: home/sonoff/stat/acsw01/POWER = ON (retained)
```

reloxx13, 2018-04-12:
their is a command with powerretain or sth, look retain up in the wiki.
but how i said, im out now, iave no mqtt server i dont know more :p

reloxx13, 2018-04-12:
#2411

dh-harald, 2018-04-12:
OC:
```
08:32:09 CMD: PowerRetain
08:32:09 MQT: home/sonoff/stat/wsw01/RESULT = {"PowerRetain":"ON"}
```

dh-harald, 2018-04-12:
I've realized, that the Sonoff Touch always writes the configuration into the Flash (``Saved to flash at FA, Count 9511, Bytes 1648``), meanwhile S20 don't

reloxx13, 2018-04-12:
check SaveData command for this

arendst, 2018-04-12:
On the Touch you have log messages set to 3 while on the S20 it is set to 2 and hence no info about flash updates.

dh-harald, 2018-04-12:
@arendst: do you have any idea about the "real" issue?

arendst, 2018-04-12:
Well looking at your initial issue it clearly states that it receives a Power off after power on most likely caused by power retain set.

As documented in answers given since then I do not go into discussing setting power retain on.

dh-harald, 2018-04-12:
@arendst: yes, I'm setting PowerRetain on, hence I'm using homeassistant...
All my experiences are weird, because with same config, S20 are working fine (they're keep turning on after reconnect), while Touches aren't

It looks, when I connects to the broker, it automatically sends an power off command:
```
2018-04-12 14:26:12.601 home/sonoff/stat/wsw01/POWER ON
---- [ mqtt restart here ] ----
2018-04-12 14:26:30.530 home/sonoff/tele/wsw01/LWT Online
2018-04-12 14:26:30.558 home/sonoff/cmnd/wsw01/POWER OFF
2018-04-12 14:26:30.583 home/sonoff/stat/wsw01/POWER ON
2018-04-12 14:26:50.897 home/sonoff/tele/wsw01/LWT Online
2018-04-12 14:26:50.942 home/sonoff/cmnd/wsw01/POWER (null)
2018-04-12 14:26:51.119 home/sonoff/stat/wsw01/RESULT {"POWER":"OFF"}
2018-04-12 14:26:51.163 home/sonoff/stat/wsw01/POWER OFF
```

Jason2866, 2018-04-12:
No offend, why do so many people use HomeAssistant? I read so many problems with and about it.
In my humble opinion for easy fast setup Domoticz is nice. For nearly everything OpenHab can be used.
I took a short look at HA and found no reason for me to use. Setting up isnt realy simple and the things i wanted to do arent easy maybe impossible... I tried OpenHab and i am happy with that choice.

kajmaj, 2018-04-13:
Jason, I am HA user. Be more specific-I use HASS.IO version.
Why? (My point of view)
- I use Raspberry Pi as server and hassio offered "one shot" installation of everything I needed, ie. mqtt server, configurator, ssh access etc.
- Initial configuration is easy, Everything works flawlessly (for me)
- Community/forum provides help with almost anything.
- I really do not have problems described here and if any, they are caused solely by my mistake
- I´ve tried Domotics and OpenHab as well but I did not feel comfortable with it. I am not saying that never ever, but so far - Hassi.io is my favorite.

dh-harald, 2018-04-14:
@arendst: I've investigated the issue, and I think, I've found the root cause. When you're using the switch with HA, it makes the command queue retained... So If I powered off a switch, the /cmnd/ topic got the last value retained into the mqtt (in this case powered off).... After reconnect, it subscribes to the /cmnd/ topic as well, and got the power off command...

So, idea: could you add a config value and functionality to the command retain as well? So if you push the physical button/through webpage, the command into the topic should be retained as well (not just the power topic)...

closed 2018-04-14 by dh-harald

carloda, 2018-10-27:
Hi, I have the same problem with sonoff touch, someone managed to solve?

ascillato, 2018-10-27:
Solution: don't use retain messages for a normal setup. Retain is meant for advance Management like for example retained variables for the mode of a thermostat. See the video from thehookup. Link in Tutorials in the wiki

carloda, 2018-10-27:
@ascillato 
thanks I solved =)