arendst/Tasmota #2425: [Request] PID controler - heater thermostat

sirLeone, 2018-04-12:
It would be great to have a possbility to control heater (e.g. water heater, electric heater) via SSR (Fotek ?) and tasmota. As a temp sensor could be used DS18x20. SSR is driven by 3-32VDC so it is not a problem to find on board.
Arduino has its own PID library so maybe it is not a huge effort to port it to esp8266.
I am sure that there are a lot of people interested in having such a functionality.

Jason2866, 2018-04-12:
Dump question what is SSR?

KPK75, 2018-04-12:
Solid State Relay?

sirLeone, 2018-04-12:
Exactly - SSR - Solid State Relay, most common (and counterfeited) is FOTEK but can can be any other with specific parameters (driven by 3-32 VDC)

Jason2866, 2018-04-13:
I dont understand why a SSR needs an change in the software. Just connect it instead of a normal relay.
Or do i miss something?

Frogmore42, 2018-04-13:
People are asking for a PID controller to be implemented in Tasmota itself.  I find it somewhat funny.  A PID controller is a commonly used system in industrial process control, but it is far from necessary, or justified, for many systems.  The far more common bang-bang controller is fine for many systems and proper use of a PID controller requires tuning and stability testing/design.  Bang-bang control is what is used by most home heating systems.  When the temperature drops below the set point, the system turns on the heater.  When it goes above the set point, it turns off the heater.

I first implemented various control strategies in a lab course 35 years ago and it was fun to watch the system respond.  Five years ago I needed a freeze protection system.  I implemented a very simple bang-bang controller.  On one part, the heater is probably 20x more than what is needed and the thermal inertia is quite large.  With 0 deadband there is still an overshoot of about 20 degrees.  In this case a nicely tuned PID controller (with at least P and I non zero) would likely offer better performance, with better defined as maintaining temperature much closer to set point.  But, since the goal was to keep the faucet from freezing, the current solution is fine.  On another part of the system, the heat input much more closely matches the need and the thermal mass of the medium is much less (difference between water and air).  On this part of the system, bang-bang keeps the temperature of the coldest part within a degree or so the set point.

Last year (before winter) I stared using Tasmota for this purchase (since my original systems started failing and were no longer available).  I use standard Tasmota, mosquito, and NodeRED.  I have a flow in NodeRED that takes the sensor readings and does the comparison and then sends a MQTT message to Tasmota to turn on the relay.  Tasmota is configured to automatically turn off the relay, which makes sure it will fail in safe condition.  I need to add a thermal limit switch as a backup (in case Tasmota or the hardware fails with the relay powered on).

If I really wanted PID control, I could add it in NodeRED.  Or, use the PID controller developed by another user.

Frogmore42, 2018-04-13:
https://github.com/arendst/Sonoff-Tasmota/wiki/PID-Control-with-Sonoff-Devices

ascillato, 2018-05-21:
Hi,

In the wiki, now is available an example of a Simple Thermostat using RULE feature (https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#7-simple-thermostat-example)

Please, check it to see if that helps you.

The control technique used there is just a simple ON/OFF with 2 set point values (upper and lower limit)

To use PID you will have to add the PID files as explained on the wiki and modify the thermostat example to use the PID commands explained inside the PIDs files.

ascillato2, 2018-06-27:
Hi, If your issue is not solved with all the links provided, please reopen it. Thanks

closed 2018-06-27 by ascillato2