arendst/Tasmota #2426: sleep freezes the devices, needs new flash: 5.12.0j

manu212, 2018-04-12:
using Sonoff Basic and Sonoff S20
-Provide the output of command ``status 0``: not possible device hangs forever

I am new to tasmota firmware. I flashed it successfully and then I tried to use the command sleep to save some power.

after using in the webinterface udner commands the command "sleep 200" or sleep 50" the devices dont work anymore. unplugging and plugging in again doesnt work. after 3 -4 times plugging in and out i can get the webinterface to show up for one html site (have to open the webinterface right after the green led lights up), but then any link leads to error mesages from the browser that the website is not available. the router confirms this since the device has no wlan connection.
I tried several times to be fast enough to enter the command "sleep 0" but it doesnt work.

I had to reflash the firmware and i won't touch this feature anymore. thre is also no warning that the device "dies".

KPK75, 2018-04-12:
Bug in 2.4.1 runtime. Use previous version.
[https://github.com/arendst/Sonoff-Tasmota/pull/1940](https://github.com/arendst/Sonoff-Tasmota/pull/1940)

manu212, 2018-04-12:
@KPK75 is it possible with OTA? Can you give me a link to a working version with sleep?

KPK75, 2018-04-12:
I managed to disable sleep via serial just right afer boot. Try

manu212, 2018-04-12:
How can i flash a working version, where do I get Version 2.4.0? Is there a binary file I can flash via OTA? Otherwise I have to remove all sonoffs to reflash agan with serial. Is this the only solution to get sleep working?

manu212, 2018-04-13:
I just downloaded the repo and folliwed the arduino manual since I know how to use Arduino and it works. No need for newest release. OTA is not working because there are only 476kb left. Flashing the minimal Version results in a reboot. Is it possible to get it Wirkung by OTA? Otherwise what do I need to download for flashing with Arduino?

curzon01, 2018-04-13:
You should use the latest release. This is an older version, it's not the developer you are using and it is compiled with the right library. I have no idea which library you are using and this point can be exclude when you use a release and not a self compiled. The release also contains minimal version.

manu212, 2018-04-13:
curl fails like the browser upload with upload buffer miscompare

manu212, 2018-04-13:
problem is still there with version 5.12.0 (official release 9th february) sleep kills the device

curzon01, 2018-04-13:
I'm using sleep since 5.11.1 and up on different Sonoff Basic, Sonoff Pow, Sonoff Dual R2, different ESP modules, and Wemos D1, no idea