arendst/Tasmota #2427: relay cycles every 45 - 50 seconds

or78rXxZ, 2018-04-12:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
sonoff basic

- [ ] Provide the output of command ``status 0``
07:23:30 MQT: stat/sonoff_b2/STATUS = {"Status":{"Module":1,"FriendlyName":["sonoff_b2"],"Topic":"sonoff_b2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
07:23:30 MQT: stat/sonoff_b2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:58:31","StartupUTC":"2018-04-12T11:24:59","Sleep":0,"BootCount":254,"SaveCount":415,"SaveAddress":"F5000"}}
07:23:30 MQT: stat/sonoff_b2/STATUS2 = {"StatusFWR":{"Version":"5.12.0i","BuildDateTime":"2018-03-26T05:57:38","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
07:23:30 MQT: stat/sonoff_b2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":3,"LogHost":"192.168.2.15","LogPort":514,"SSId":["dzc_ntA61Fvgbhit",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
07:23:30 MQT: stat/sonoff_b2/STATUS4 = {"StatusMEM":{"ProgramSize":518,"Free":484,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
07:23:30 MQT: stat/sonoff_b2/STATUS5 = {"StatusNET":{"Hostname":"sonoff-b2","IPAddress":"192.168.2.26","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"104.223.91.194","Mac":"EC:FA:BC:86:AB:8A","Webserver":2,"WifiConfig":3}}
07:23:30 MQT: stat/sonoff_b2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.16","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_86AB8A","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
07:23:30 MQT: stat/sonoff_b2/STATUS7 = {"StatusTIM":{"UTC":"Thu Apr 12 12:23:30 2018","Local":"Thu Apr 12 07:23:30 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-5}}
07:23:30 MQT: stat/sonoff_b2/STATUS10 = {"StatusSNS":{"Time":"2018-04-12T07:23:30"}}
07:23:30 MQT: stat/sonoff_b2/STATUS11 = {"StatusSTS":{"Time":"2018-04-12T07:23:30","Uptime":"0T00:58:31","Vcc":3.471,"POWER":"ON","Wifi":{"AP":1,"SSId":"dzc_ntA61Fvgbhit","RSSI":44,"APMac":"A0:F3:C1:1A:2B:24"}}}

- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

I have expect this has been addressed but I'll be damned if I can find the answer. I just got several sonoff basic and connected several lights. When they are on and they lose connection with the access point every 45 - 50 seconds the relay cycles off and on very quickly. This happens with all 4 I have installed.

Jason2866, 2018-04-12:
Core":"2_4_1","SDK  isnt a good one. Use 2.4.0 or even better for Tasmota 2_3_0/1.5.3

or78rXxZ, 2018-04-13:
Thanks. Is that a bug in  2.4.1 or just an incompatibility issue?

Jason2866, 2018-04-13:
It is a bug in 2.4.1

Jason2866, 2018-04-13:
The question is why do the loose Wifi connection?
Is the problem the device or is it the Wifi network.
Just try the prebuild version and not your own, maybe your compiled versions are all faulty

or78rXxZ, 2018-04-13:
I am using the prebuilt file I got from https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.12.0/sonoff.bin
The problem occurs on all 4 sonoff basic

Jason2866, 2018-04-13:
Post a log file when it cycles. Maybe there is some info that helps

or78rXxZ, 2018-04-13:
What is the best way to get a log?
Also I got 2 sonoff s20 today and they behave the same.
Am I the only one that this has ever happened to?

Jason2866, 2018-04-13:
Do you flashed your image (post 1) or the prebuilt 5.12.0 ?
To get a log just copy and paste the output from the web console

or78rXxZ, 2018-04-13:
I am using the prebuilt file I got from https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.12.0/sonoff.bin
The cycling only occurs when the device is not able to connect to the access point.
The web console is not available

Jason2866, 2018-04-13:
The reason is WHY is it not able to connect to your network!
Try to change your cryptic SSID dzc_ntA61Fvgbhit in a simple one 
Check your network settings in general

or78rXxZ, 2018-04-13:
The reason it is not able to connect to my access point is because I shutdown the access point to test the behavior of the device when it is not able to connect to an access point.
The question is what is wrong with the firmware?

Jason2866, 2018-04-13:
Without a log nothing can be analyzed

or78rXxZ, 2018-04-13:
I just started using tasmota about a week ago. Tell me how to get a log and I will.

Jason2866, 2018-04-15:
Make sure your wifi is working? Do the device gets a IP address from your DHPC Server (most cases the router). If yes try to reach the webconsole. If it cycyles about 45sec you should see something on the web console. The same place where you get your status 0 infos. Just copy this and paste it here