arendst/Tasmota #2429: Problem connecting to WLAN with multiple access points carrying the same ssid

mikeww1, 2018-04-12:
Make sure these boxes are checked before submitting your issue - Thank you!

- [  ] What hardware you are using
Sonoff basic and T1 EU 2

I have multiple access points in my house and all are configured with the same ssid and pwd.
As soon as one of the tasmota devices can receive two of them it connects and disconnects in a loop.
If I switch off one and the switch can only see a single AP it connects. When I switch on the other again it disconnects. I solved the issue by creating a guest wlan with a unique ssid. Now everything works.
I just wanted to document that behaviour to safe you some time if you run into that issue.

Jason2866, 2018-04-12:
Hmmm, i have 4 Accespoints (all use the same SSID) in my house and no problem with my >10 Tasmota devices .

ssirkakriss, 2018-04-12:
I had a similar experience. However it turned out to be a misconfigured AP. One of my AP's was on a different channel then the others and causing the closest Sonoff-Tasmota's to drop off.  I only have 3 AP's and 5-6 Sonoff-Tasmota's ATM.

curzon01, 2018-04-13:
as suggested before:
- be sure all AP has same SSid, same PW, same channel.
- if you only use one SSid, do not configure the seconds SSid with Tasmota - keep the second Wifi configuration to a invalid one
- provide your `Status 0` (to see your config e. g. WifiMode etc.)

wongnam, 2018-04-13:
should not use same channel for this case, it will leading these AP will conflict each other.
i am using 2 AP for 15 Sonoff device without any problem.
My configuration:
- Same SSID
- Same User Name and Password
- Same Authenticate and Encryption method eg: WPA2 and AES

davidelang, 2018-04-13:
On Thu, 12 Apr 2018, Norbert Richter wrote:

> as suggested before:
> - be sure all AP has same SSid, same PW, same channel.

the APs do not need to be on the same channel (and in fact, probably should not 
be), but they do need to have the same SSID and password, and a few other 
things.

1. don't use DHCP on each AP, have one DHCP for the entire network
2. have the APs bridge the wifi to a network rather than route from the wifi 
network

I run the wifi network for the Scale conference, we have ~120 APs all on the 
same SSID to support over 3k devices

https://www.usenix.org/conference/lisa12/technical-sessions/presentation/lang_david_wireless
https://www.usenix.org/publications/login/april-2013-volume-38-number-2/wireless-means-radio

ssirkakriss, 2018-04-13:
I had a opportunity to revisit why I was able to connect when all AP's used same channel. and not when on different channels.

I removed one (home made) AP from the network because it has a tendency to cause some problems. The remaining two AP's (UniFi AP about 45' apart) where set on channel one (1). I restarted all devices flashed with Tasmota and all reconnected, with no problems.

I then changed one AP to channel six (the other remained on channel one) and rebooted all Tasmota  devises. Most devices worked, except one. This device is smack in the middle range of the two AP's.  It is a wemos mini 1M version it is running the release version 5.12.0. It was very slow and appeared to be reestablishing a connection all the time then dropping off. Sorry no log this devise in my house thermostat and the serial port is not easily accessible. 

I then changed one AP's channel from six to eleven. I rebooted the problematic device. It appears to be working as expected. I can log in, faster. And I do not see console messages about reconnecting.

I reestablished the third (home made) AP and set that to channel six. Rechecked all the device and it appeared to have no ill effects. This AP is far from the others and I'm fairly sure the range does not overlap with the other two AP's

All Tasmota are happy and healthy with the AP's on different channels now. 

However it is puzzling when AP-1 on channel one (1) and AP-2 on channel six (6). The one Tasmota device in the middle had connection issues. But not when AP-1 set to channel one (1) and AP-2 set to channel eleven (11).

Just my observation.

Jason2866, 2018-04-14:
I use 4 AP only with same SSID. The APs (all Business AP from LANCOM or Bintec-elmeg) use channel autoselect so the are definitly on different channels and no problems withy even one Tasmota device ( > 10). I am sure your wifi setup is the problem. There are to less issues that indicates that is a problem with Tasmota software... Just my 2 cents