arendst/Tasmota #2431: Philips HUE App tries to update Sonoff Basic w/Hue Bridge emulation

PhelanH, 2018-04-12:
I'm using some Sonoff Basic Switches with HUE Bridge Emulation with an Amazon Alexa.

This works as expected. However I have a lot of "normal" HUE Devices. And using the Philip HUE App.

If the App discovers the Sonoff devices it tries to update the "Bridge" When I try to chancel the update the HUE App is not usable anymore (till the Sonoff "HUE Bridges" are updated or disconnected) Since we can obviously not update the HUE bridge emulation via the Hue App, I was wondering if we could have a input field for the version in the web backend like it's done here:
[https://github.com/bwssytems/ha-bridge/wiki/HA-Bridge-FAQs#q-the-hue-app-says-i-need-to-update-the-firmware-version-how-do-i-do-that](https://github.com/bwssytems/ha-bridge/wiki/HA-Bridge-FAQs#q-the-hue-app-says-i-need-to-update-the-firmware-version-how-do-i-do-that)

This way we could change the reported version of the Sonoff Hue Bridge, so the HUE App would complain about an outdated bridge.

Make sure these boxes are checked before submitting your issue - Thank you!

- [X ] What hardware you are using => Sonoff Basic w/Tasmota 5.11.1
- [X ] Provide the output of command ``status 0``
18:06:17 CMD: status 0
18:06:17 RSL: STATUS = {"Status":{"Module":1,"FriendlyName":"Badezimmerspiegel","Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:06:17 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":0,"Sleep":0,"BootCount":30,"SaveCount":69,"SaveAddress":"F8000"}}
18:06:17 RSL: STATUS2 = {"StatusFWR":{"Version":"5.11.1","BuildDateTime":"2018.01.07 21:51:30","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:06:17 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"domus1","LogPort":514,"SSId1":"Pretty Fly for a WiFi","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000001"}}
18:06:17 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
18:06:17 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-1663","IPAddress":"192.168.2.71","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"DC:4F:22:09:66:7F","Webserver":2,"WifiConfig":1}}
18:06:17 RSL: STATUS7 = {"StatusTIM":{"UTC":"Thu Apr 12 17:06:17 2018","Local":"Thu Apr 12 18:06:17 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
18:06:17 RSL: STATUS10 = {"StatusSNS":{"Time":"2018.04.12 18:06:17"}}
18:06:17 RSL: STATUS11 = {"StatusSTS":{"Time":"2018.04.12 18:06:17","Uptime":0,"Vcc":3.227,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Pretty Fly for a WiFi","RSSI":50,"APMac":"70:8B:CD:AE:38:00"}}}
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

austinlg96, 2018-04-18:
I'm running 5.12.0 on a Sonoff Basic and I was able to bypass the update screen by modifying xplg_wemohue.ino. Changing the swversion on line 493 to "[1802201122](https://www2.meethue.com/en-us/support/release-notes/bridge)" seems to do it. However, I just installed Tasmota for the first time, so I don't know how the Hue app is supposed to respond now. (I don't have any actual Hue devices either.)  

The only thing that I can do in the Hue app is fail at assigning the Sonoff to a room. Before the update issue, did it ask to assign it to a room? Are you supposed to be able to have any control of the Sonoff in the Hue app or is it only for Alexa et al compatibility? #854 included changing the reported api level in the fix for #698 but I don't know if changing it to "[1.24.0](https://www.developers.meethue.com/documentation/changelog)" would actually make something incompatible.

PhelanH, 2018-04-18:
It's mostly for the reason to still use the HUE App for ones original HUE Devices. As soon as HUE Bridge emulation by a Tasmato powered Devices is detected by the App, you basicly have only the option to Update the Tasmato HUE Bridge (which will fail) or the HUE App from Philips goes in some sort of lookdown until all bridges it detects are updated. At least this is the case on android devices (don't know about IOS)

A workaround is to cut Power to all Tasmato devices that are emulating a HUE bridge while using the HUE App, but this cause a major inconvinience.

Idealy we could get the current HUE Bridge version from some external source, but I could live with having to Update all the Tasmato devices in my network by hand via the Webinterface. A Plus would be the posibillity to set the HUE Bridge version via some sort of API so one could batch update his Tasmato devices.

cmavr8, 2018-05-20:
I'm experiencing the same issue with version 5.14.0a. The Hue emulation is useless to me, as the sonoff device cannot be adopted in the Philips Hue app. Am I doing something wrong?

Even if I allow the app to attempt to "upgrade" the sonoff, it will fail and result in the app being completely useless; genuine Hue lights cannot be controlled any more. On Android I need to wipe the app's data and rediscover the genuine Hue bridge to restore functionality.

closed 2018-05-20 by arendst

cmavr8, 2018-05-21:
That worked! Thanks! The Hue app now allows me to add the sonoff as a new bridge. 

Unfortunately though, it doesn't allow me to add a scene and thus control the device. 
As far as I understand, this is expected, as described in https://github.com/arendst/Sonoff-Tasmota/issues/999#issuecomment-337494172. 

I guess the Hue emulation is not intended to be used with the Hue app but perhaps Alexa or something else. Am I right or am I missing something? If that's the case, we should make it a bit more clear in the documentation, where Hue emulation is mentioned, so people don't get their hopes up.

erichegit, 2018-12-26:
I also would like to know - can a Tasmota flashed device be discovered by a hardware HUE-Bridge=?

altelch, 2018-12-26:
Hi,

no, Tasmota emulates a HUE-Bridge and not a zigbee-device. A huebridge 
would detect zigbee devices and not other hue-bridges I'd guess.

Greetings,

      Altelch.

PR #2788 (2018-05-20): Update to HUE Bridge V1 Software Version
Should fix #2431