arendst/Tasmota #2432: Sonoff Basic with wrong PulseTime after MQTT command

pplucky, 2018-04-12:
**Tasmota Version:**
5.11.1

**Hardware:**
Sonoff Basic

**Output of command Status 0: (non relevant info redacted)**
`{"Status":{"Module":1,"FriendlyName":"Sonoff-comando","Topic":"sonoff-comando","ButtonTopic":"sonoff-comando","Power":0,"PowerOnState":3,"LedState":3,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":1,"Sleep":0,"BootCount":123,"SaveCount":263,"SaveAddress":"F5000"},"StatusFWR":{"Version":"5.11.1","BuildDateTime":"2018-01-07T21:51:09","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"},"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":4,"LogHost":"172.xx.xx.15","LogPort":514,"SSId1":"MySSID1","SSId2":"MySSID2","TelePeriod":300,"SetOption":"55800009"},"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3},"StatusNET":{"Hostname":"sonoff-comando-5014","IPAddress":"172.xx.xx.61","Gateway":"172.xx.xx.1","Subnetmask":"255.255.255.0","DNSServer":"172.xx.xx.1","Mac":"AA:BB:CC:DD:EE:FF","Webserver":2,"WifiConfig":2},"StatusMQT":{"MqttHost":"172.xx.xx.15","MqttPort":1883,"MqttClientMask":"sonoff-comando","MqttClient":"sonoff-comando","MqttUser":"mosquitto","MAX_PACKET_SIZE":512,"KEEPALIVE":15},"StatusTIM":{"UTC":"Thu Apr 12 17:04:12 2018","Local":"Thu Apr 12 18:04:12 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1},"StatusSNS":{"Time":"2018-04-12T18:04:12"},"StatusSTS":{"Time":"2018-04-12T18:04:12","Uptime":1,"Vcc":3.262,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MySSID1","RSSI":100,"APMac":"FF:EE:DD:CC:BB:AA"}}}
`

**Output of command Pulsetime1:**
`{"PulseTime1":11}`

**Log level:**
'4. More debug'

**HomeAssistant switch config:**
```
switch:
  - platform: mqtt
    name: 'Portão Garagem Prédio'
    command_topic: 'cmnd/sonoff-comando/POWER'
    availability_topic: "tele/sonoff-comando/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    state_topic: 'none'
    optimistic: false
    payload_on: "ON"

  - platform: command_line
    switches:
      portao_garagem_predio_d:
        friendly_name: 'Portão Garagem Prédio D'
        command_on: '/usr/bin/curl -s http://172.16.23.61/cm?cmnd=Power%20On'
        command_off: '/usr/bin/curl -s http://172.16.23.61/cm?cmnd=Power%20Off'
```

**Issue description:**
PulseTime provided by Tasmota is not the same when sending an 'ON' command via mqtt or via URL.

If I publish "ON" to MQTT topic 'cmnd/sonoff-comando/POWER' (switch portao_garagem_predio in Home Assistant), PulseTime seems not to be fulfilled as defined (1.1 seconds), while when calling URL for Power On (switch portao_garagem_predio_d in Home Assistant), it seems OK.

**More debug log:**
```
2018-04-12T18:09:40.663120+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/RESULT = {"POWER":"ON"}  <-- Power On
2018-04-12T18:09:40.668695+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/POWER = ON
2018-04-12T18:09:41.733534+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/RESULT = {"POWER":"OFF"} <-- PulseTime
2018-04-12T18:09:41.738221+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/POWER = OFF

Pulse Time of 1,070414 seconds (OK)

2018-04-12T18:10:11.582594+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/RESULT = {"POWER":"ON"}  <-- Power On
2018-04-12T18:10:11.587896+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/POWER = ON               
2018-04-12T18:10:12.653462+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/RESULT = {"POWER":"OFF"} <-- PulseTime
2018-04-12T18:10:12.658972+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/POWER = OFF

Pulse Time of 1,076378 seconds (OK)

2018-04-12T18:10:45.473966+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/RESULT = {"POWER":"ON"}  <-- MQTT ON
2018-04-12T18:10:45.480656+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/POWER = ON
2018-04-12T18:10:45.794904+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/RESULT = {"POWER":"OFF"} <-- PulseTime
2018-04-12T18:10:45.799109+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/POWER = OFF

Pulse Time of 0,325143 seconds (NOK)
```

These issues are rendering the usage of MQTT command to be useless, as the time between ON and OFF (controlled by PulseTime), is not enough to fulfil the purpose.

Is there anything that can be done to improve/fix it or is this a known issue (although I could not find it from searching issues in GitHub)?

Please let me know if there's anything else I can do to help figure this out.

curzon01, 2018-04-13:
[v5.12.0](https://github.com/arendst/Sonoff-Tasmota/releases/tag/v5.12.0) already released

davidelang, 2018-04-13:
see the troubleshooting section of the wiki, specifically the poweronstate 
retained message page, this sounds like you have a pulsetime messaage being 
retained.

pplucky, 2018-04-14:
For the time being, did 2 things:
- Upgrade to 5.12.0 (after this, PulseTime seems bigger when power turned on via MQTT, but still below 1,1 seconds - which is still the one used when power is turned on via URL);
- Changed PowerOnState from 3 to 0 (PowerRetain was already 0) --> to be tested further.

I'll test in the next couple of days and come back with the results.

@davidelang In the [troubleshooting Wiki section](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting), found nothing on PowerOnState, just on [Usage](https://github.com/arendst/Sonoff-Tasmota/wiki/PowerOnState-Configuration). I suppose that's what you meant (and based on that, I changed the setting for PowerOnState).

pplucky, 2018-04-15:
I've tested multiple times and PulseTime when power is turned on via MQTT is definitely wrong, as compared when power is turned on via URL.

Check this example (1st test wth MQTT, 2nd test with URL):

```
2018-04-15T17:58:00.835593+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/RESULT = {"POWER":"ON"}  <-- Power On
2018-04-15T17:58:00.841785+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/POWER = ON
2018-04-15T17:58:01.014865+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/RESULT = {"POWER":"OFF"} <-- PulseTime
2018-04-15T17:58:01.020138+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/POWER = OFF

Pulse Time of 0,184545 seconds (NOK)

2018-04-15T17:58:05.017825+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/RESULT = {"POWER":"ON"}  <-- Power On
2018-04-15T17:58:05.024836+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/POWER = ON
2018-04-15T17:58:06.099257+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/RESULT = {"POWER":"OFF"} <-- PulseTime
2018-04-15T17:58:06.105075+01:00 sonoff-comando-5014 ESP-MQT: stat/sonoff-comando/POWER = OFF

Pulse Time of 1,08725 seconds (OK)
```

Any other ideas?

pplucky, 2018-04-16:
@curzon01 I have always downloaded from release repository. Upgraded to 5.12.0 and result is exactly the same.

It's pretty odd that when using URL Power ON it consistently behaves well (PulseTime = +/ 1.1sec), issue seems to be only with the MQTT command. 
It should also not be an MQTT server issue either, because PulseTime should be handled by Tasmota after the Power ON being sent to MQTT server.

Any ideas where to look at?

Thanks in advance.