arendst/Tasmota #2433: Flashed Sonoff Basic - Web server does not respond

Aileron79, 2018-04-12:
Hi!
I've managed to successfully flash Tasmota on my Sonoff Basic. I followed the instructions from 
DrZzs' Youtube video. The device successfully requests an IP address, but connecting to the web server only worked the very first time, and only for the index page. Since then, all requests to port 80 just time out:

`#  nmap -sS -Pn 192.168.0.237

Starting Nmap 6.47 ( http://nmap.org ) at 2018-04-12 19:47 CEST
Nmap scan report for sonoff_01-5423 (192.168.0.237)
Host is up (0.0016s latency).
Not shown: 999 closed ports
PORT   STATE SERVICE
80/tcp open  http
MAC Address: 5C:CF:7F:xx:xx:xx (Unknown)
`

Connecting to port 80 works, but I don't get an answer at all:

`telnet 192.168.0.237 80
Trying 192.168.0.237...
Connected to 192.168.0.237.
Escape character is '^]'.
GET / HTTP/1.1

`

Any idea what the issue could be? The device definitely has been flashed without any problems. I saw the Tasmota web GUI only once, I was not able to successfully reconnect. Plugging out and in again the Sonoff does not work. Any ideas?

curzon01, 2018-04-13:
FW version? Status 0?

Assuming your compile Tasmota by your own and you are using lib 2.4.0 or 2.4.1? 
First try [latest binary release](https://github.com/arendst/Sonoff-Tasmota/releases/latest)

pascalsaul, 2018-04-13:
Looks like: https://github.com/arendst/Sonoff-Tasmota/issues/2319#issuecomment-380964405

curzon01, 2018-04-18:
pls close if solved

closed 2018-06-05 by ascillato