arendst/Tasmota #2434: RF Bridge registration of event on Water Leakage Sensor Digoo

ddossev, 2018-04-12:
Hi,

I need a solution to limit the time RF Bridge waits for a signal to end. Below is description of the issue I have.

Yesterday I received a cheap Digoo water leak sensor [(link)](https://www.banggood.com/Digoo-DG-HOSA-433MHz-Wireless-Water-Leakage-Alarm-Water-Level-Detector-for-Home-Security-Guarding-p-1178109.html?cur_warehouse=CN) 

I tried to link it to Sonoff Bridge that is flashed with Tasmota 5.12 (details follow).

The code of event is captured in a strange way. When a leakage is detected a lamp on the sensor is becoming red. It stops to be red at the moment the sensor is taken out of the water. At the same moment (taking out the sensor of water) RF Bridge captures the code. (I do not want to understand that my apartment was flooded but now the water is away :))

The device has also two buttons that send code to RF Bridge. The signal is received from RF Bridge at the moment the buttons are not pressed.

As an alternative I started RF Key learn mode. When I simulate leakage RF Bridge beeps without able to see anything in the console. The key is registered only I stop the simulation. If I keep it for long time I receive several beeps and at the end error message that learning was not completed.

My conclusion is that RF Bridge is registering event when a signal is received and waits for the time to stop receiving signal.

I do not have such issue with Sonoff PIR that sends signal for a very short time. 
I do have the same problem with a Door Sensor having a separate button - event is not registered until I release the button.

Can somebody share a solution how I can limit the time RF Bridge is waiting to receive the whole signal?

Thanks in advance

- [ ] What hardware you are using - **Sonoff Bridge**

- [ ] Provide the output of command ``status 0``:

`20:26:20 CMD:  status 0
20:26:20 RSL: Group 0, Index 1, Command STATUS, Data 0
20:26:20 MQT: stat/sonoffrf01/STATUS = {"Status":{"Module":25,"FriendlyName":"Sonoff","Topic":"sonoffrf01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
20:26:20 MQT: stat/sonoffrf01/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoffrf01.ino.bin","Uptime":"0T00:10:28","Sleep":0,"BootCount":6,"SaveCount":13,"SaveAddress":"FB000"}}
20:26:20 MQT: stat/sonoffrf01/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-03-13T15:33:10","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:26:20 MQT: stat/sonoffrf01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"XXXX","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000009"}}
20:26:20 MQT: stat/sonoffrf01/STATUS4 = {"StatusMEM":{"ProgramSize":497,"Free":440,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
20:26:20 MQT: stat/sonoffrf01/STATUS5 = {"StatusNET":{"Hostname":"sonoffrf01-6008","IPAddress":"192.168.0.XXX","Gateway":"192.168.0.YYY","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.ZZZ","Mac":"XXXXXXXX","Webserver":2,"WifiConfig":3}}
20:26:20 MQT: stat/sonoffrf01/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.105","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_82D778","MqttUser":"YYYYYY","MAX_PACKET_SIZE":1024,"KEEPALIVE":15}}
20:26:20 MQT: stat/sonoffrf01/STATUS7 = {"StatusTIM":{"UTC":"Thu Apr 12 19:26:20 2018","Local":"Thu Apr 12 20:26:20 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
20:26:20 MQT: stat/sonoffrf01/STATUS10 = {"StatusSNS":{"Time":"2018-04-12T20:26:20"}}
20:26:20 MQT: stat/sonoffrf01/STATUS11 = {"StatusSTS":{"Time":"2018-04-12T20:26:20","Uptime":"0T00:10:28","Vcc":3.159,"Wifi":{"AP":1,"SSId":"XXXX","RSSI":76,"APMac":"ZZZZZZ"}}}`

Jason2866, 2018-04-13:
If the RfBridge receives a signal that is correctly decoded it sends it immediatly via mqqt and you can see it on the webconsole.
So in your case i see 2 possibilitys
Code is not correct recognized or decoded or the sensor is really a "funny" one...

ddossev, 2018-04-14:
What I explained earlier the code is received only when the sensor stops to send signal. 

I assume that there is no problem with RF recognizing the signal. 

I will check with a different transmitter what happens if e.g. a remote button is hold constantly. 

Regards
Delyan

Jason2866, 2018-04-14:
I think the RF Bridge has problems decoding the signal. The receiver of the RF Bridge is a "good" one.
But the signal is decoded with a EFM8BB1 in Rf Bridge and this knows only 24 bit codes.
So if transmitters use a unusual protocol the rf bridge tries to receive and after a timout it fails.
There is a project to build an alternative firmware... see https://github.com/Portisch/RF-Bridge-EFM8BB1

ddossev, 2018-04-15:
Hi,
Thanks for your prompt replies. 

I am not sure if try are capturing my proposal.

Signal is captured. The difference is when the RF recognizes it. 

If I press for 0.5 second I receive notification in 0.5 seconds. If I press for 30'seconds I receive notification in 30 seconds. 

I saw that those sensors are working with Rflink I will try some time with it. 

Regards 
Delyan 

Regards 
Delyan