arendst/Tasmota #2436: I flash Sonoff SC but it doesn't work

paulo9999, 2018-04-12:
Hello, 

I flashed the last version 5.12 into Sonoff SC (I connected back the TX jumper after flashing). But when I turn on, the led blinks and I cannot see it at the network. 

What does it mean the following statement: (I don't know how to set the baudrate to 19200)

After flashing you need to set the baudrate to 19200. Don't forget to reconnect the TX jumper after flashing ;)

Can anyone help me? thank you.

Jason2866, 2018-04-13:
Prebuild Image or a self compiled with your wifi settings?
if precompiled you have to configure your wifi see wiki
https://github.com/arendst/Sonoff-Tasmota/wiki/Initial-Configuration

Jason2866, 2018-04-13:
If you get in the web console you have to enter: 
baudrate 19200

paulo9999, 2018-04-13:
I

paulo9999, 2018-04-13:
I didn't get to the web console, and yes I had setup my wifi at the code before compiling.

paulo9999, 2018-04-17:
I've compiled the 5.12v code and sent to the ESP with success. 

BUT when I connect at the outlet, the led keeps blinking and I cannot do anything else. I changed the WIFI settings with my SSID and password, so the Sonoff can connect to the network, but I can't see the device connected. Is there any way to debug this problem??? Thank you.

Jason2866, 2018-04-17:
Maybe wrong settings are stored in the device. Try
https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#config-problems-can-cause-boot-loops-items-set-in-user_config-to-not-appear-etc

paulo9999, 2018-04-18:
Finally I could connect the device. Still don't know exactly what occurred.... 
at the console I typed the command but I think that it didn't recognize it... 

01:56:18 CMD: baudrate 19200
01:56:18 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}

paulo9999, 2018-04-18:
At the end it worked! At the console I saw the automatic change of the baudrate, but I didn't change anything. 

Thank you @Jason2866 !

closed 2018-04-18 by paulo9999