arendst/Tasmota #2438: Not retained Alarms messages

ornolio, 2018-04-13:
I’ve 1 “Sonoff RF Bridge 433" running Tasmota 12.0 + "Sonoff DW1 Door Window Alarm Sensor" setted at RfKey 7. On DW1 alarmed, the Bridge sends the following payload:  

11:31:00 MQT: tele/bridge-1/RESULT = {"RfReceived":{"Sync":14700,"Low":490,"High":1460,"Data":"26F341","RfKey":7}}

Perfectly reported by mosquitto_sub on the server as: 
{"RfReceived":{"Sync":14700,"Low":490,"High":1460,"Data":"26F341","RfKey":7}}

However that and any other payload from the Bridge is not RETAINED; on the contrary any payload sent by server using mosquito_pub with option “r” is correctly copied and  retained.

I tried on Bridge console some setup commands like:  sensorretain, powerretain but unsuccessful.

As for semplicity I don't use any home automation integration application running continuosly, but the MQTT Dash by Routix APP on mobile phone, I loose every alarm occurred when the APP is not running.

Is there any suggestion/help to let the Bridge going to retain that payload, as described above ?

Thank You in advance
Piero

- [ ] Provide the output of command ``status 0``
12:33:55 MQT: stat/bridge-1/STATUS = {"Status":{"Module":25,"FriendlyName":"Sonoff","Topic":"bridge-1","ButtonTopic":"bridge-1","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":1,"PowerRetain":1}}
12:33:55 MQT: stat/bridge-1/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.77:80/api/arduino/sonoff.ino.bin","Uptime":"0T15:49:05","Sleep":0,"BootCount":15,"SaveCount":43,"SaveAddress":"F9000"}}
12:33:55 MQT: stat/bridge-1/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:33:55 MQT: stat/bridge-1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"192.168.1.77","LogPort":514,"SSId1":"TIM-24084959","SSId2":"TIMB-24084959","TelePeriod":300,"SetOption":"00000669"}}
12:33:55 MQT: stat/bridge-1/STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
12:33:55 MQT: stat/bridge-1/STATUS5 = {"StatusNET":{"Hostname":"bridge-1-6534","IPAddress":"192.168.1.225","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:8F:39:86","Webserver":2,"WifiConfig":3}}
12:33:55 MQT: stat/bridge-1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.77","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8F3986","MqttUser":"ornolio","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:33:55 MQT: stat/bridge-1/STATUS7 = {"StatusTIM":{"UTC":"Fri Apr 13 11:33:55 2018","Local":"Fri Apr 13 12:33:55 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
12:33:55 MQT: stat/bridge-1/STATUS10 = {"StatusSNS":{"Time":"2018-04-13T12:33:55"}}
12:33:55 MQT: stat/bridge-1/STATUS11 = {"StatusSTS":{"Time":"2018-04-13T12:33:55","Uptime":"0T15:49:05","Vcc":3.182,"Wifi":{"AP":1,"SSId":"TIM-24084959","RSSI":100,"APMac":"10:13:31:70:95:F3"}}}

ornolio, 2018-04-14:
Thank you Jason for your promptly reply. 
Really it should be unnecessary to store the questioned payload on the device which of course is short of memory (unless customer had replaced the memory with a larger one :-) ) ; it would be enought add the byte "R" to the mqtt command sent to the broker by the device, of course under some existing flag set to on (sensorretain ?),  
Reason is that most likely in order to keep low batteries consumption, either DW1s (Door and Window sensors) or PIRs send only alarms occourred, but do not repeat continuosly their status on/off to the Bridge device and in turn to the MQTT server broker. 
I believe the alarms of these sensors should be avalable to any newvly mqtt client any time it subscribes the corresponding topic; in my case a mobile APP MQTT Dash by Routix which could be off when the sensor alarm events occours.
Currently those "unretained" Sonoff Bridge alarms are only available to the subscribers up and running at the same time the alarms occurred, if not they're lost.
I know that a complex alternatives would be to add an home automation integration application or  running on the server a customized service subscribing Bridge 433 topics and then "re-publish" those payloads with option Retain set on, but that would be like to shoot a "mosquito" with a cannon :)).   

Thank You for your patience if you read everything.
Best Regards.
P.

P.S. here is a reference on MQTT retained messages by HIVEMQ:
https://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages
substaining my above comments on:
a)where are retained payload stored
b)what to do by the perspective of a developer
c)which is the retained payload useful for

QUOTE by HIVEMQ at above link.

A retained message is a normal MQTT message with the retained flag set to true. The broker will store the last retained message and the corresponding QoS for that topic....

Sending a retained message from the perspective of a developer is quite simple and straight-forward. You just need to set the retained flag of a MQTT publish message to true. Each client library typically provides an easy way to do that.

A retained message makes sense, when newly connected subscribers should receive messages immediately and shouldn’t have to wait until a publishing client sends the next message. 

UNQUOTE

Jason2866, 2018-04-14:
Ok, i do understand... But i believe @arendst will not implement. But hey asking doesnt matter.
I came from a position like you. I wanted to control smart devices (first step power outlets)
and dont like the idea to give the control out of my house.... On the other side i dont want invest hundreds of euros. So i searched and found the sonoff devices with firmware Tasmota. But just controling every device via web isnt user friendly at all. The WAF factor is null! So i looked for a reach featured and easy setup solution HA system. So i installed OpenHab on a RPi3. Sd Card Image. Setup was done in a hour.

ornolio, 2018-04-14:
Tank You again Jonson, I understood your position. 

Actually it is neither a question of RPi3 cost nor complexity. First because the MQTT broker I use currently sits on a RPi3, second because OpenHab come on SD image, so I'm expecting it would be only a matter of configuration and setup.

My point is that I've to disrupt and rebuild all my applications. Just to give you an idea of the work to do,  please consider I've:
1)at Home 
18 Sonoff Touch, 3 of them mounting DS18B20 Temp. Sensors, 1 Sonoff Basic, 1 Sonof Bridge 433 driving 2xDW1+1xPIR
This central Rpi3 itself runs the MQTT broker and in addition I publish by phyton customized scripts some additional telemetry using DHT22 sensor for Temp+Hum + BMP180 for Barometric pressure on its GPIOs.
This RPI3 is duplicated by a second one, just for matter of availability.

2)at a remote  cottage far away 10 Km from home in the hills, there is another RPI collecting temp+Hum from 3xDHT22 + Barometric pressure from another BMP180, and alarms from 2xDW1 + 1xPIR linked to another Sonoff Bridge RF433. All these telemetry are published remotely on the above MQTT central broker at home on 4G net.

3)Last at a remote sail boat in the port of my city (Varazze at 30Km from Genoa) a third RPI is collecting on Raspi GPIO Temp & Hum by a DHT22 + Barometric pressure by a BMP180 + wind direction and intensity on a serial RS232 interface to Raspi  GPIO, again publishing by 4G Net using customized python scripts, this telemetry on the central RPI at Home which, by the way, has encyption SSL/TLS certbot for RASPI.

As you can understand, it's hard to decide to rebuild all this stuff for reason of 1 flag, particularly thinking that at the end of the story this would be not useful only for me, but for many others which will love Tasmota firmware as I do.

Anyway, thank You again for your attention and let sit and wait for Arendst opinion, if any.

Best Regards.
P

ornolio, 2018-04-14:
this shows the 3 location on the MQTT Dash APP by Routix
[image]

this shows the topics at home part 1
[image]
this shows the topics at home part 2
[image]
this shows the topics at cottage
[image]
and last the topics at sailboat
[image]

ornolio, 2018-04-19:
I solved this issue with an external working around. 

I implemented a python script on the same server of the broker (RPI3 B)  which,  using paho-mqtt python companion, which subscripts the RF BRIDGE 433 alarms topic "tele/bridge-1/RESULT", getting the payload and replacing it with a new one before publishing it on a new topic, really used by the MQTT Dash by Routix.

When an alarm is published on "tele/bridge-1/RESULT" topic by the Tasmota flashed RF Bridge 433, I get the payload and change the item  "RfKey":"<key number>" to "RfKey":"ON", publishing adding  Qos=0 and Retain flag "1" this new payload on a new topic "tele/bridge-1/RESULTx" where "x" postfix is the RfKey No. get from the first topic "tele/bridge-1/RESULT" .

All the raised alarm are then stored (retained) on their new topics, each one  corresponding to the raised RF Bridge 433 key. 

Later I'm able to read the alarms even if the MQTT Dash APP by Routix APP was switched off or the phone itself is switched off when the alarm occurred. 

I can also reset the alarm using the MQTT Dash APP by Routix "ON TAP" function, publishing "OFF" Qos=0 Retain=1 on the "tele/bridge-1/RESULTx".

Now working to show the alarm time-stamps on the MQTT Dash APP by Routix and also to create an alarm log for all the configured keys of the Tasmota flashed RF Bridge 433. 

here is the python prototype
#### per sottoscrivere i messaggi dei sensori e ri-pubblicarli con retain = true ######
import paho.mqtt.client as mqtt #import the client1
import time
############

def on_message(client, userdata, message):
        print("message received " ,str(message.payload.decode("utf-8")))
        print("message topic=",message.topic)
        print("message qos=",message.qos)
        print("message retain flag=",message.retain)
        if   ('"RfKey":6' in message.payload):
                print("message payload conts RfKey:6",message.payload)  #studio Mola
                client.publish("tele/bridge-1/RESULT6","ON",0,retain=True)
#       elif ('"RfKey":7' in message.payload): 
#               print("message payload conts RfKey:7",message.payload)  #poggiolo Mola
#               client.publish("tele/bridge-1/RESULT7","ON",0,retain=True)
        elif ('"RfKey":8' in message.payload):
                print("message payload conts RfKey:8",message.payload)  #ingresso Mola
                client.publish("tele/bridge-1/RESULT8","ON",0,retain=True)
########################################

print("creating new instance")
client = mqtt.Client("P1") #create new instance
print("connecting to broker")
client.username_pw_set("mymqttbrokeruser", "mymqttbrokerpassword")
client.connect("mybroker address",1883,60) #connect to broker
client.loop_start()
client.on_message=on_message #attach function to callback
client.subscribe("tele/bridge-1/RESULT")

time.sleep(4) # wait

client.loop_forever()

=======================================
here is the sub topic of the alarm key no. 6 on MQTT Dash by Routix APP:
tele/bridge/RESULT6
it receive "ON" or "OFF" which are recognized by the APP which shows a bell icon red and blinking if "ON", whithe, barred and not blinking if "OFF" 

here is the MQTT DASH by Routix APP "ON TAP" script to acknowledge the alarm on key 6:
app.publish("tele/bridge-1/RESULT6","OFF",true,0);

closed 2018-04-19 by ornolio