arendst/Tasmota #2440: Sonoff DEV - any connected Switch to any GPIO toggles Power

kajmaj, 2018-04-14:
There are 7 GPIOs where is possible to connect any sensor, powered by 3.3 or 5 Volts.
I need to use 4 switches+a few more sensors.
During testing I recognised that sensors are fine, but all switches toggle Power. (i.e. any change of any switch from ON-OFF / OFF-ON toggle POWER.
Does not matter on GPIO (1,3,4,5,12,13).
I did not tested GPIO 14, as I need it for anything else. 
DEV if flashed by 5.12.0j 
Is is correct behavior?

Short example of log, switches are connected to GPIO1,3 and SerialLog OFF.

`06:30:49 MQT: tele/DEV/STATE = {"Time":"2018-04-14T06:30:49","Uptime":"0T00:14:53","Vcc":3.462,"Wifi":{"AP":1,"SSId":"MyNET","RSSI":100,"APMac":"F4:F2:6D:9B:94:A2"}}
06:30:49 MQT: tele/DEV/SENSOR = {"Time":"2018-04-14T06:30:49","Switch1":"OFF","Switch2":"OFF"}
06:30:55 MQT: stat/DEV/RESULT = {"POWER":"OFF"}
06:30:55 MQT: stat/DEV/POWER = OFF
06:30:59 MQT: tele/DEV/STATE = {"Time":"2018-04-14T06:30:59","Uptime":"0T00:15:03","Vcc":3.462,"Wifi":{"AP":1,"SSId":"MyNET","RSSI":100,"APMac":"F4:F2:6D:9B:94:A2"}}
06:30:59 MQT: tele/DEV/SENSOR = {"Time":"2018-04-14T06:30:59","Switch1":"OFF","Switch2":"ON"}
06:31:02 MQT: stat/DEV/RESULT = {"POWER":"ON"}
06:31:02 MQT: stat/DEV/POWER = ON
06:31:09 MQT: tele/DEV/STATE = {"Time":"2018-04-14T06:31:09","Uptime":"0T00:15:13","Vcc":3.462,"Wifi":{"AP":1,"SSId":"MyNET","RSSI":100,"APMac":"F4:F2:6D:9B:94:A2"}}
06:31:09 MQT: tele/DEV/SENSOR = {"Time":"2018-04-14T06:31:09","Switch1":"ON","Switch2":"ON"}
06:31:14 MQT: stat/DEV/RESULT = {"POWER":"OFF"}
06:31:14 MQT: stat/DEV/POWER = OFF`

Status 0:
`06:53:37 CMD: Status 0
06:53:37 MQT: stat/DEV/STATUS = {"Status":{"Module":19,"FriendlyName":[],"Topic":"DEV","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
06:53:37 MQT: stat/DEV/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Power on","Uptime":"0T00:00:39","StartupUTC":"2018-04-14T05:52:58","Sleep":0,"BootCount":133,"SaveCount":534,"SaveAddress":"F9000"}}
06:53:37 MQT: stat/DEV/STATUS2 = {"StatusFWR":{"Version":"5.12.0j","BuildDateTime":"2018-04-07T18:16:57","Boot":4,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
06:53:37 MQT: stat/DEV/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HugoNET",""],"TelePeriod":10,"SetOption":["00000009","55818000"]}}
06:53:37 MQT: stat/DEV/STATUS4 = {"StatusMEM":{"ProgramSize":504,"Free":496,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
06:53:37 MQT: stat/DEV/STATUS5 = {"StatusNET":{"Hostname":"DEV-0219","IPAddress":"192.168.1.59","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:92:40:DB","Webserver":2,"WifiConfig":3}}
06:53:37 MQT: stat/DEV/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.107","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9240DB","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
06:53:37 MQT: stat/DEV/STATUS7 = {"StatusTIM":{"UTC":"Sat Apr 14 05:53:37 2018","Local":"Sat Apr 14 06:53:37 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
06:53:37 MQT: stat/DEV/STATUS10 = {"StatusSNS":{"Time":"2018-04-14T06:53:37","Switch1":"OFF","Switch2":"OFF"}}
06:53:37 MQT: stat/DEV/STATUS11 = {"StatusSTS":{"Time":"2018-04-14T06:53:37","Uptime":"0T00:00:39","Vcc":3.462,"Wifi":{"AP":1,"SSId":"MyNET","RSSI":100,"APMac":"F4:F2:6D:9B:94:A2"}}}`

reloxx13, 2018-04-14:
check switchtopic in wiki

kajmaj, 2018-04-14:
``
19:34:10 CMD: SwitchTopic 0
19:34:10 MQT: stat/DEV/RESULT = {"SwitchTopic":""}
19:34:14 MQT: stat/DEV/RESULT = {"POWER":"ON"}
19:34:14 MQT: stat/DEV/POWER = ON
19:34:14 MQT: tele/DEV/STATE = {"Time":"2018-04-14T19:34:14","Uptime":"0T00:07:03","Vcc":3.462,"Wifi":{"AP":1,"SSId":"MyNET","RSSI":100,"APMac":"F4:F2:6D:9B:94:A2"}}
19:34:14 MQT: tele/DEV/SENSOR = {"Time":"2018-04-14T19:34:14","Switch1":"ON","Switch2":"OFF","DS18B20":{"Temperature":25.6},"TempUnit":"C"}
19:34:17 MQT: stat/DEV/RESULT = {"POWER":"OFF"}
19:34:17 MQT: stat/DEV/POWER = OFF
``
From my point of view those two are redundand and not needed if we are speaking solely about sensors:
`19:34:17 MQT: stat/DEV/RESULT = {"POWER":"OFF"}
19:34:17 MQT: stat/DEV/POWER = OFF`

Frogmore42, 2018-04-16:
How exactly do you have things wired and configured?
Your first log shows a case where the sensor values for the two switches are in different states.  But, 3 seconds later a command appears to come to change the state of one of them.

So, what is the exact behavior you are seeing?
Is there a delay between the relays or LEDs switching?
Which one switches first?
What home automation do you have hooked up?

I don't use multiple switches, but I know other people do with success.