arendst/Tasmota #2442: Usage of Wire library.

Koepel, 2018-04-14:
The Wire library is used in the wrong way in some files.

File "Sonoff-Tasmota/sonoff/support.ino", function "I2cValidRead()", there is an extra Wire.endTransmission() at the end, that should not be there.

File "Sonoff-Tasmota/lib/Joba_Tsl2561/src/Tsl2561.cpp", the "readByte()" and "readWord()" have extra Wire.endTranmission() in the return.

File "Sonoff-Tasmota/lib/Mutichannel_Gas_Sensor/src/MutichannelGasSensor.cpp",  function "readData()" has a timeout, but there is no need to wait for something after a Wire.requestFrom().

The file "Sonoff-Tasmota/lib/I2Cdevlib-Core/I2Cdev.cpp" has a lot of bugs and mismatches. Changing the lower level TWI functions has made it worse.

Explanation: [Common-mistakes](https://github.com/Koepel/How-to-use-the-Arduino-Wire-library/wiki/Common-mistakes)

arendst, 2018-04-14:
As I don't control the libs I can only reply to your I2cValidRead() concern.

Although you might be right it's nice to know more people are making mistakes while the code still seems to work.

Thnx for pointing it out.

Koepel, 2018-04-14:
There will probably be extra I2C transactions on the bus without any meaning, and code that makes no sense. Some background information: http://arduinoforum.nl/viewtopic.php?f=12&t=2929

joba-1, 2018-05-11:
Lib version 2.0.6 has the fix. I tested it with 5.13.1a, patched with my lib version now for some time and it (still) works just fine.

It is a cosmetical fix that might reduce i2c bus traffic slightly, but no functional differences: no need to hurry...

closed 2018-09-04 by ascillato2

Koepel, 2019-05-14:
The function `I2cValidRead()` does still not use the Wire library in a valid way, yet this issue has been closed. I have checked all four branches.

arendst, 2019-05-14:
How about this:
```
bool I2cValidRead(uint8_t addr, uint8_t reg, uint8_t size)
{
  uint8_t retry = I2C_RETRY_COUNTER;
  bool status = false;

  i2c_buffer = 0;
  while (!status && retry) {
    Wire.beginTransmission(addr);                       // start transmission to device
    Wire.write(reg);                                    // sends register address to read from
    if (0 == Wire.endTransmission(false)) {             // Try to become I2C Master, send data and collect bytes, keep master status for next request...
      Wire.requestFrom((int)addr, (int)size);           // send data n-bytes read
      if (Wire.available() == size) {
        for (uint8_t i = 0; i < size; i++) {
          i2c_buffer = i2c_buffer << 8 | Wire.read();   // receive DATA
        }
        status = true;
      }
    }
    retry--;
  }
  return status;
}
```

Koepel, 2019-05-14:
That is correct use of the Wire library. Thank your for fixing it.

Is it really necessary ? What good does it do ?
A glitch on SDA causing wrong data is not detected, the I2C protocol has no checksum.
The Arduino Wire library can detect only a few problems, and those problems are not even likely to occur. Maybe all the extra code can prevent 1% of the bus problems.

My opinion about error checking is on this page: [Check for bus errors ?](https://github.com/Koepel/How-to-use-the-Arduino-Wire-library/wiki/Check-for-bus-errors-%3F).