arendst/Tasmota #2445: Extend Sonoff RF Bridge 433 to store more than 16 keys

beebodo, 2018-04-14:
Hi!
I wonder why, according to Sonoff original firmware, Tasmota only supports 16 keys to store RF codes for Sonoff RF Bridge 433. To my mind there should be no hardware limitation for this and it should be possible to extend the firmware to use more keys. Am I wrong? Did anybody have any experience how to extend the module that way?

Leseratte10, 2018-04-14:
If I remember correctly, the RF stuff is not handled by the ESP8266 chip which Tasmota runs on, but by a dedicated 433-RF chip which just sends a button press signal to the ESP when it receives a remote signal.

gheesung, 2018-04-16:
There is no need to learn the RF code and use the key. You can directly send the code using RfCode and the bridge will out the RF signal. For RF receive, the bridge will return the code received in the "tele" message. Your receiving application only needs to listen to the queue and interpret the code.

mientki, 2018-11-05:
I have door contact sensors that emit 3 different codes: open, closed and tampered. If the Sonoff RF Bridge only supports 16 keys, then I can add only 5 sensors. Is it possible to add more codes? I need to add 10 sensors x 3 codes = 30 rfkeys. Is it possible?

KptnKMan, 2019-03-13:
Still interested in this issue here.

ascillato, 2019-03-13:
Please, re read the comment from @gheesung:

> There is no need to learn the RF code and use the key. You can directly send the code using RfCode and the bridge will out the RF signal. For RF receive, the bridge will return the code received in [...]

Beside that, you can also use the rfcodes in rules for doing any behaviour you want.

There is no need to add more buttons in the webui as rfkey30.

ascillato, 2019-03-13:
For more information, please check the wiki and for support, you can ask in the Tasmota support chat. Thanks

KptnKMan, 2019-03-13:
My apologies, I checked the wiki and I saw other closed issue #4013
I'm trying to find where in the wiki I can specify a MQTT message for the same.

Is this the correct part of the Wiki?
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#how-to-use-commands
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#sonoff-rf-bridge
Thanks.