arendst/Tasmota #2446: 5.12.0k messed up?

kskenyon, 2018-04-14:
Something is wrong with 5.12.0k built with ATOM on a Mac.  Previous versions are still working on other devices.

POWER is continuously sent to MQTT and the web interface becomes unresponsive.  It restarts eventually.  I've reinstalled and reconfigured software several different ways and it's the same.

15:25:07 MQT: stat/TH16/STATUS = {"Status":{"Module":1,"FriendlyName":["TH16"],"Topic":"TH16","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":7,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
15:25:07 MQT: stat/TH16/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:28","StartupUTC":"2018-04-14T19:13:39","Sleep":0,"BootCount":8,"SaveCount":19,"SaveAddress":"F9000"}}
15:25:07 MQT: stat/TH16/STATUS2 = {"StatusFWR":{"Version":"5.12.0k","BuildDateTime":"2018-04-13T18:35:05","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
15:25:07 MQT: stat/TH16/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["netis_2.4G_E7D2C0","netis_2.4G_E7D2C0"],"TelePeriod":300,"SetOption":["00008109","55818000"]}}
15:25:07 MQT: stat/TH16/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
15:25:07 MQT: stat/TH16/STATUS5 = {"StatusNET":{"Hostname":"TH16-4397","IPAddress":"10.0.1.124","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"75.75.75.75","Mac":"5C:CF:7F:40:D1:2D","Webserver":2,"WifiConfig":4}}
15:25:07 MQT: stat/TH16/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.32","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_40D12D","MqttUser":"pi3","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:25:07 MQT: stat/TH16/STATUS7 = {"StatusTIM":{"UTC":"Sat Apr 14 19:25:07 2018","Local":"Sat Apr 14 15:25:07 2018","StartDST":"Sun Mar 04 02:00:00 2018","EndDST":"Sun Nov 04 03:00:00 2018","Timezone":99,"Sunrise":"12:13","Sunset":"25:51"}}
15:25:07 MQT: stat/TH16/STATUS10 = {"StatusSNS":{"Time":"2018-04-14T15:25:07"}}
15:25:07 MQT: stat/TH16/STATUS11 = {"StatusSTS":{"Time":"2018-04-14T15:25:07","Uptime":"0T00:11:28","Vcc":3.306,"POWER":"ON","Wifi":{"AP":1,"SSId":"netis_2.4G_E7D2C0","RSSI":98,"APMac":"08:10:78:E7:D2:C6"}}}
15:25:07 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:07 MQT: stat/TH16/POWER = ON
15:25:07 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:07 MQT: stat/TH16/POWER = ON
15:25:07 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:07 MQT: stat/TH16/POWER = ON
15:25:07 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:07 MQT: stat/TH16/POWER = ON
15:25:07 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:07 MQT: stat/TH16/POWER = ON
15:25:07 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:07 MQT: stat/TH16/POWER = ON
15:25:07 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:07 MQT: stat/TH16/POWER = ON
15:25:07 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:07 MQT: stat/TH16/POWER = ON
15:25:07 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:07 MQT: stat/TH16/POWER = ON
15:25:07 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:07 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:08 MQT: stat/TH16/RESULT = {"POWER":"ON"}
15:25:08 MQT: stat/TH16/POWER = ON
15:25:13 RSL: stat/TH16/RESULT = {"POWER":"ON"}
and so on until it restarts...

Make sure these boxes are checked before submitting your issue - Thank you!

- [x] What hardware you are using
- [x ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

curzon01, 2018-04-15:
Try version 5.12.0k compiled with lib 2.3.0

Jason2866, 2018-04-15:
Maybe only cosmetic, i would use a DNS Server in your local network.
In most cases your router does this.