arendst/Tasmota #2447: Add Random Function for the Timer

FabianHM, 2018-04-14:
Hello,
Is it possible for the Timer to include a random Function as a presence simulation?

The calling and ending of the random function should be done via the normal commands.

Thanks and greetings
Fabian

species5618, 2018-04-15:
great idea, for security purposes, 
keeps people watching lights on and off , an thinking its a timer
some commercial devices i have seen offer +/- **X** minutes

renne, 2018-04-16:
Homematic e.g. offers a random offset for timers.

Geitde, 2018-04-16:
yeah, such option would be great and just needs one byte per timer where the value would be stored.

The offset should go in both directions. So a value of 15 would result in random timers between -15 and +15 minutes.

renne, 2018-04-18:
`random time = (setpoint - offset) + (random number % (offset * 2))`

FabianHM, 2018-04-19:
Thank you Theo for the realization of my idea :-)
Unfortunately, I do not understand without explanation the function of the Random Timer. The following entry did not lead to success. My goal is that at sunset until sunrise, the random circuit takes place in selectable intervals.

[image]

arendst, 2018-04-19:
The general feeling about the random functionality is that it takes place only once around the set time at + or - max 15 minutes. This allows for a random switch to keep the burglars out (they think).

You have to fix your goal using rules and rule#timers.

localhost61, 2018-04-20:
@arendst , in the Timer Json, the (undocumented) "Window" parameter is not yet parsed. So one cannot change the random field of Timers from the CLI.
`15:20:00 CMD: timer1`
`15:20:00 RSL: RESULT = {"Timer1":{"Arm":1,"Mode":0,"Time":"15:10","Window":0,"Days":"1111111","Repeat":0,"Output":1,"Action":3}}`
`15:20:15 CMD: timer1 {"Arm":1,"Mode":0,"Time":"15:10","Window":0,"Days":"1111111","Repeat":0,"Output":1,"Action":3}`
`15:20:15 RSL: RESULT = {"Timer1":"Invalid JSON"}`
`15:20:25 CMD: timer1 {"Arm":1,"Mode":0,"Time":"15:23","Days":"1111111","Repeat":0,"Output":1,"Action":3}`
`15:20:25 RSL: RESULT = {"Timer1":{"Arm":1,"Mode":0,"Time":"15:23","Window":0,"Days":"1111111","Repeat":1,"Output":1,"Action":3}}`

localhost61, 2018-04-21:
Ok! I thought is was not yet implemented. The issue was not were I thought :-).

closed 2018-06-02 by ascillato