arendst/Tasmota #2448: LwIP variant "v1.4 Prebuilt" not available in Arduino IDE

dweston, 2018-04-15:
- [ x] What hardware you are using
MacOS 10.10.5
Arduino IDE 1.8.5 installed as portable config using guidance here:
http://forum.arduino.cc/index.php?topic=539551.msg3678998#msg3678998
Tasmota install followed as per:
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

Reached the "Configure Arduino IDE" stage and all OK except there is no 'v1.4 Prebuilt' option available in the "IwIP Variant:" drop down menu.
Only options available are:
v2 Lower Memory
v2 Higher Bandwidth
v1.4 Higher Bandwidth
v1.4 Compile from source

Have I missed something?

- [? ] Provide the output of command ``status 0``
Didn't get that far.
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
n/a
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst
n/a

alton76, 2018-04-15:
I’m assuming you are using ESP2.4 built

Select V2 lover Memory

dweston, 2018-04-15:
Yes I'm using the ASP2.4.1 build.

Thank you. I will proceed with "v2 Lower Bandwidth" selected.

arendst, 2018-04-15:
Just updated the wiki with the relevant information https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

closed 2018-04-15 by arendst

dweston, 2018-04-15:
Interesting that after selecting the "v2 Lower Bandwidth" LwIP variant option, as per the advice above, when I proceeded to the "Compile Tasmota" action the complied failed - 'exit status 1'. While I considered what to try next I took a look back at the 'IwIP Variant:' options menu and the content had entirely changed from the original:
"v2 Lower Memory
v2 Higher Bandwidth
v1.4 Higher Bandwidth
v1.4 Compile from source"

to

"v2 Prebuilt (MSS=536)
v2 Prebuilt (MSS=1460, unstable)
v1.4 Prebuilt
v1.4 Compile from source"

Given earlier advice I'd noted in this issues list, I selected 'v1.4 Prebuilt' and tried to re-compile.
The compile failed again 

I think the most relevant elements of the compile error messages - which appear identical to the first compile attempt - are as follows: 

``Arduino: 1.8.5 (Mac OS X), Board: "Generic ESP8266 Module, Serial, 80 MHz, nodemcu, 26 MHz, 40MHz, DOUT, 1M (no SPIFFS), v1.4 Prebuilt, Disabled, None, 115200"
.......
/libraries/SPI/SPI.cpp.o" "/var/folders/kv/n5vj2lv54bs9blbz41m3nqnc0000gs/T/arduino_build_181118/arduino.ar" -lhal -lphy -lpp -lnet80211 -llwip_gcc -lwpa -lcrypto -lmain -lwps -laxtls -lespnow -lsmartconfig -lairkiss -lmesh -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group  "-L/var/folders/kv/n5vj2lv54bs9blbz41m3nqnc0000gs/T/arduino_build_181118"
/Applications/Tasmota.app/Contents/Java/portable/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot find -lmesh
collect2: error: ld returned 1 exit status
........
exit status 1
Error compiling for board Generic ESP8266 Module.``

arendst, 2018-04-15:
Your file paths may be too long using kv/n5vj2lv54bs9blbz41m3nqnc0000gs ?

arendst, 2018-04-15:
Mesh is part of the ESP8266 Arduino library. I suggest you reinstall the library as you also mentioned you suddenly moved from 2.4.1 options (v2 lower memory) to 2.4.0 options (v1.4 prebuilt).

It seems your installation is not complete.

dweston, 2018-04-15:
I uninstalled the ESP8266 Board version 2.4.1 and reverted to ESP8266 Board version 2.4.0.

I checked the IwIP Variant: menu options and 'v1.4 Prebuilt' was still present as an option.

The compile has now completed without error.

dweston, 2018-04-15:
Looks like a successful install:
[Sonoff_POW_upload.PDF](https://github.com/arendst/Sonoff-Tasmota/files/1912978/Sonoff_POW_upload.PDF)

meingraham, 2018-04-21:
Replacing the platform file with the 2.4.1 version resolved the compilation error. Thanks!

gsbad, 2018-05-02:
Same problem here