arendst/Tasmota #2451: External switch on GPIO 14

MrJoki007, 2018-04-15:
I want to connect my wall switch to GPIO 14 on my Sonoff Basic to toggle it.
I connected it between Ground and the fifth pin (GPIO 14).
In the webinterface i changed GPIO 14 to 09 Switch 1.
If i connect or disconnect ground and GPIO 14 nothing happens.
Did i do something wrong?

localhost61, 2018-04-15:
Did you look at the log on the console on the webserver? If nothing appear, go and configure logging on Level2 for WebLog
(I currently only have Dual R2 with latest 5.12.0l)

MrJoki007, 2018-04-16:
I set WebLog to 2 but it does not show anything when i connect the both contacts

reloxx13, 2018-04-16:
- Restart device
- Try with shorter/other cables
- Check your soldering/resolder pins

MrJoki007, 2018-04-16:
Restarted it a few times
Already switched cables (short jumpers)
The soldering should be decent

MrJoki007, 2018-04-25:
What do you mean?

MrJoki007, 2018-04-25:
Just tried it out. Sometimes it works sometimes not.

MrJoki007, 2018-04-29:
It worked but it was on switchmode 3 then i changed it to switchmode 0 and it stopped: 
status 0: https://pastebin.com/sgrxp1AW

titomontes, 2018-04-30:
What do you mean with "it stopped"?
Does changing switchmode to 0 have any effect on the console log when shorting GPIO14-GND? (also, try shorting it directly with one single cable from below, looking at the soldering points)
Please paste that

closed 2018-06-29 by stale[bot]