arendst/Tasmota #2453: AP Ssid names not appearing to compile correctly

dweston, 2018-04-16:
- [X] What hardware you are using
MacOSX 10.10.5
PlatformIO
Sonoff POW
Serial upload

Followed set-up tutorial as per:
https://www.youtube.com/watch?v=n4MDRm2yAJg
except with the addition that the file 'platformio.ini' was altered to uncomment line 14:
`env_default = sonoff`

- [X] Provide the output of command ``status 0``
`00:00:31 CMD: status 0
00:00:31 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic
":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:00:31 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domu
s1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:30","StartupUTC":"
","Sleep":0,"BootCount":415,"SaveCount":861,"SaveAddress":"F7000"}}
00:00:31 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.12.0l","BuildDateTime":"2018-04-16T11:14:04","Boo
t":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
00:00:31 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPor
t":514,"SSId":["indebuurt1","indebuurt2"],"TelePeriod":300,"SetOption":["00000009","55818000"]}}
00:00:31 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":527,"Free":476,"Heap":23,"ProgramFlashSize":1024
,"FlashSize":4096,"FlashMode":3}}
00:00:31 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5879","IPAddress":"0.0.0.0","Gateway":"192.

168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"BC:DD:C2:EA:96:F7","Webserver":2,"Wifi
Config":1}}
00:00:31 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"domus1","MqttPort":1883,"MqttClientMask":"DVES_%06
X","MqttClient":"DVES_EA96F7","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:31 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:31 1970","Local":"Thu Jan 01 00:00:31
1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","
Sunset":"16:03"}}
00:00:31 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:31"}}
00:00:31 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:31","Uptime":"0T00:00:30","Vcc":3.32
4,"POWER":"OFF","Wifi":{"AP":2,"SSId":"indebuurt2","RSSI":100,"APMac":"FF:FF:FF:FF:FF:FF"}}}`

As may be seen in the screen shot of the PlatformIO IDE, while the SSID settings have been edited into 'user_config.h' the Ssid names being reported in the Serial Monitor appear to be the default names - hence probably why the device fails to connect to the network.

[image]

Is this a bug or situation normal.

dweston, 2018-04-16:
Yep - my first flash was the plain original code as I understood it was still possible to access the device via its default webserver address. Unfortunately I could not manage to trigger that state nor find the device broadcasting its own WiFi presence. The advice at https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage referred to flashing & blinking LED feedback, which just isn't happening with the configuration I've ended up with. Definitely no LED state changes apart from relay on/off. I can't tell you how many hours and options I tried until I finally worked out how to interpret the feedback from the Serial Monitor sufficiently to capture the device off my iPhone. Traps for early players I dare say.

May I add my awe at comprehending what Theo and associates have managed to achieve in 1Mb of memory.

reloxx13, 2018-04-16:
https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#config-problems-can-cause-boot-loops-items-set-in-user_config-to-not-appear-etc

dweston, 2018-04-17:
Much obliged - thank you.

I'm still baffled by the absence of button functionality LED flashing feedback.