arendst/Tasmota #2454: Sonoff Pow continuous measurement

papa-romeo, 2018-04-16:
I use a Sonoff-POW to measure a refrigerator and a freezer.
Because it would have fatal consequences if, for example, switch off the POW, I bridged the relay outputs.
So the devices are constantly supplied with the mains voltage, regardless of the switch position (relay position) of the POW's.
The old Sonoff MQTT-OTA software (v.3.2.13) did not care how the switches in the Sonoffs stood.
The measured values were recorded and transmitted at "ON" as well as at "OFF".
This no longer works with the new TASMOTA software.
Here, the switch must be forced to "ON", so that the measured values are recorded and transmitted.
Can this be changed somewhere in the sketch?

Leseratte10, 2018-04-16:
According to the command list https://github.com/arendst/Sonoff-Tasmota/wiki/Commands, `SetOption21 1` should be what you are looking for (just enter that into the web console).

papa-romeo, 2018-04-16:
Hallo Leseratte10, herzlichen Dank. Das war´s. Funktioniert perfekt. Ich habe mir die Commands wohl angesehen, aber diese Einstellung wohl überlesen.

Hello Leseratte10, thank you. That's it. Works perfectly. I have the Commands well looked at, but probably overread this setting.

closed 2018-04-18 by arendst